espressif/arduino-esp32 #5491: ESP32-CAM deep sleep wakeup problem with arduino 1.06

lunadm, 2021-08-03:
* Development Board: ESP32-CAM
* Module or chip used: ESP32-S
* IDF version: Arduino 1.06 esptool v3.0-dev
* Operating System: Mac
* Power Supply: [USB/battery]
* Chip is ESP32-D0WD (revision 1)

I’ve recently migrated from 1.04 to 1.06 stable arduino 
(if I revert back to 1.04 the problem doesn’t exist)
And I’m encountering a random problem waking up from deep sleep 
The ESP32-CAM successfully wakes up but randomly hangs at boot

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

- I cant reproduce the problem- it happens randomly
It can go for hours/days  or it could happen after the first sleep
It happens regardless of the power source 
I’ve also tried different 5v boosters  (currently using the 5v VIN with a Pololu TPS63060 c800ma)
I’ve tried different esp32-cam boards 
If I remove all the RTC variables the problem doesn’t appear (or its a coincidence) 

sleep code:
esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_TIMER); //disable previous light sleep timer
pinMode(4, OUTPUT);
digitalWrite(4, LOW); 
rtc_gpio_hold_en(GPIO_NUM_4);
esp_sleep_enable_ext0_wakeup(GPIO_NUM_13, 0);
esp_deep_sleep_start();

me-no-dev, 2021-08-10:
can you try that with the latest 2.0.0-RC1? The issue is not arduino related but rather in the underlying ESP-IDF

lunadm, 2021-08-13:
thanks - I've updated to 2.0 RC1 - will try

(fyi  I encountered
"ESPAsyncWebServer" undefined reference to `mbedtls_md5_starts'
copied over the src code/fixes from espasyncwebserver and compiles successfully.)

lunadm, 2021-08-19:
Hi I've tried 2.0 RC1 and the waking up problem occurred twice - I'll keep testing

closed 2021-08-31 by lunadm