espressif/arduino-esp32 #5493: Touch INterrupt Problem on ARDUINO IDE

KaruF, 2021-08-04:
Hi,

I want to use touch interrupt feature on esp32. Soü ı opened example code from library: TouchInterrupt. When ı compile and upload to esp32 ı saw just _ESP32 Touch Interrupt Test_ text on serial port screen. ı touched T0 and T3 pins but esp32 has never went into interrupt functions. I tried that code multiple computer but result was same. 

I use Arduino IDE 1.8.15 version with Nodemcu V1 ESP32 Devkit.

Note that touchRead function works very well. Only problem is about touch interrupt.

Code:

```
/*
This is an example how to use Touch Intrrerupts
The bigger the threshold, the more sensible is the touch
*/

int threshold = 40;
bool touch1detected = false;
bool touch2detected = false;

void gotTouch1(){
 touch1detected = true;
}

void gotTouch2(){
 touch2detected = true;
}

void setup() {
  Serial.begin(115200);
  delay(1000); // give me time to bring up serial monitor
  Serial.println("ESP32 Touch Interrupt Test");
  touchAttachInterrupt(T0, gotTouch1, threshold);
  touchAttachInterrupt(T3, gotTouch2, threshold);
}

void loop(){
  if(touch1detected){
    touch1detected = false;
    Serial.println("Touch 1 detected");
  }
  if(touch2detected){
    touch2detected = false;
    Serial.println("Touch 2 detected");
  }
}
```

KaruF, 2021-08-09:
Anyone!!!

me-no-dev, 2021-08-10:
you did not fill the issue template. There is missing information :) What version of the ESP32-Arduino are you using?

KaruF, 2021-08-10:
@me-no-dev  I tried with 2 version of Arduino IDE. Arduino IDE 1.8.15 and Arduino IDE 1.8.5. Both of them doesn't work for me.
I installed latest version of esp32-arduino file from github.

me-no-dev, 2021-08-11:
@KaruF I can confirm the issue. Was able to get around it by calling `touchRead(pin)` for each used pin in the loop. Will look into it

PR #5527 (2021-08-12): Fixes Touchpad Interrupt
Fixes #5493 

## Summary
PR #4996 has broken Touch Interrupt functionality by removing a single line of code used to set a Register.

## Impact
This PR fixes issue #5493 by reverting the removal of necessary code as described above.
commit 2af8cc34 (2021-08-12): Fixes Touchpad Interrupt (#5527)

Fixes #5493 

## Summary
PR #4996 has broken Touch Interrupt functionality by removing a single line of code used to set a Register.

## Impact
This PR fixes issue #5493 by reverting the removal of necessary code as described above.
PR #6194 (2022-02-03): Touch Sensor IDF Refactoring
## Summary
Touch Sensor refactoring to be based on IDF 4.4.
Adds support to ESP32S2 and future ESP32S3.

#### Adds some new APIs:
#### For all chips:
```void touchAttachInterruptArg(uint8_t pin, void (*userFunc)(void*), void *arg, uint32_t threshold);```
This function allows the user to add and pass a void* parameter to the ISR user callback.

```void touchDetachInterrupt(uint8_t pin);```
This function detaches ISR call back for the touch pad pin.

#### Only ESP32 chip
```void touchInterruptSetThresholdDirection(bool mustbeLower);```
This function allows the user t [... cut]
commit 0ea485e5 (2022-02-03): Touch Sensor IDF Refactoring (#6194)

Summary

Touch Sensor refactoring to be based on IDF 4.4.
Adds support to ESP32S2 and future ESP32S3.

Adds some new APIs:

For all chips:

void touchAttachInterruptArg(uint8_t pin, void (*userFunc)(void*), void *arg, uint32_t threshold);
This function allows the user to add and pass a void* parameter to the ISR user callback.

void touchDetachInterrupt(uint8_t pin);
This function detaches ISR call back for the touch pad pin.

Only ESP32 chip

void touchInterruptSetThresholdDirection(bool mustbeLower);
This function allows the user to set if the ISR callba [... cut]
closed 2022-02-03 by me-no-dev