espressif/arduino-esp32 #5494: ESP32-C3 : GLITCH_RTC_RST  if WiFi on

al1fch, 2021-08-04:
### Hardware:

|<B>Board</B>    TTGO  T-OI PLus (ESP32-C3)
|<B>Version/Date</B>  1.1
|<B>IDE name</B> Arduino 1.8.15  + ESP32 Core v 2.0.0-rc1
|<B>Flash Frequency</B>   40 MHz
|<B>PSRAM enabled</B>  no
|<B>Upload Speed</B>  115200
|<B>Computer OS</B>  Ubuntu

### Description:
Cyclic reboot (GLITCH_RTC_RST) as soon as WiFI starts
Clean power, no glitch seen on scope for 3.3V rail

### Sketch:  
 Any WiFI exemple for ESP32-C3 from Arduino IDE , even WiFIScan

### Debug Messages:
`07:20:54.278 -> ESP-ROM:esp32c3-api1-20210207
07:20:54.278 -> Build:Feb  7 2021
07:20:54.278 -> rst:0x1 (POWEI (244) esp_image: segment 4: paddr=000b5ff8 vaddr=403863d0 size=069d8h ( 27096) load
07:20:54.278 -> I (250) esp_image: segment 5: paddr=000bc9d8 vaddr=50000000 size=00010h (    16) load
07:20:54.278 -> I (254) boot: Loaded app from partition at offset 0x10000
07:20:54.278 -> I (254) boot: Disabling RNG early entropy source...
07:20:54.278 -> [   276][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
07:20:54.278 -> [   300][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 0 - WIFI_READY
07:20:54.420 -> [   437][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 2 - STA_START
07:20:54.520 -> Setup doneESP-ROM:esp32c3-api1-20210207
07:20:54.553 -> Build:Feb  7 2021
07:20:54.553 -> rst:0x13 (GLITCH_RTC_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
07:20:54.553 -> SPIWP:0xee
07:20:54.553 -> mode:DIO, clock div:2
07:20:54.553 -> load:0x3fcd6100,len:0x142c
07:20:54.553 -> load:0x403ce000,len:0xbe8
07:20:54.553 -> load:0x403d0000,len:0x292c
07:20:54.553 -> SHA-256 comparison failed:
07:20:54.553 -> Calculated: a2104353fca28c61965343021ba4ec5ec5448cc15b2cdfcf01199ea0614dc0c5
07:20:54.553 -> Expected: 1b38c41cb11a47abdb84862e1e0376a06b2ed31c486f188e95161eb8c7db7a4e
07:20:54.586 -> Attempting to boot anyway...
07:20:54.586 -> entry 0x403ce000
07:20:54.586 -> I (49) boot: ESP-IDF v4.4-dev-2030-gd93887f9f-dirty 2nd stage bootloader
07:20:54.586 -> I (49) boot: compile time 01:17:48
07:20:54.586 -> I (49) boot: chip revision: 3
07:20:54.586 -> I (50) boot_comm: chip revision: 3, min. bootloader chip revision: 0
07:20:54.586 -> I (65) qio_mode: Enabling QIO for flash chip WinBond
07:20:54.586 -> I (65) boot.esp32c3: SPI Speed      : 40MHz
07:20:54.619 -> I (65) boot.esp32c3: SPI Mode       : QIO
07:20:54.619 -> I (68) boot.esp32c3: SPI Flash Size : 4MB
07:20:54.619 -> I (72) boot: Enabling RNG early entropy source...
07:20:54.619 -> I (77) boot: Partition Table:
07:20:54.619 -> I (79) boot: ## Label            Usage          Type ST Offset   Length
07:20:54.619 -> I (86) boot:  0 nvs              WiFi data        01 02 00009000 00005000
07:20:54.619 -> I (92) boot:  1 otadata          OTA data         01 00 0000e000 00002000
07:20:54.653 -> I (99) boot:  2 app0             OTA app          00 10 00010000 00200000
07:20:54.653 -> I (105) boot:  3 spiffs           Unknown data     01 82 00210000 001f0000
07:20:54.653 -> I (112) boot: End of partition table
07:20:54.653 -> I (115) boot_comm: chip revision: 3, min. application chip revision: 0
07:20:54.653 -> I (121) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=16950h ( 92496) map
07:20:54.743 -> I (144) esp_image: segment 1: paddr=00026978 vaddr=3fc8ce00 size=032c8h ( 13000) load
07:20:54.743 -> I (147) esp_image: segment 2: paddr=00029c48 vaddr=40380000 size=063d0h ( 25552) load
07:20:54.743 -> I (153) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=85fd0h (548816) map
07:20:54.765 -> I (245) esp_image: segment 4: paddr=000b5ff8 vaddr=403863d0 size=069d8h ( 27096) load
07:20:54.765 -> I (250) esp_image: segment 5: paddr=000bc9d8 vaddr=50000000 size=00010h (    16) load
07:20:54.789 -> I (255) boot: Loaded app from partition at offset 0x10000
07:20:54.789 -> I (255) boot: Disabling RNG early entropy source...
07:20:54.818 -> [   277][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
07:20:54.818 -> [   301][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 0 - WIFI_READY
07:20:54.951 -> [   432][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 2 - STA_START
07:20:55.051 -> Setup doneESP-ROM:esp32c3-api1-20210207
07:20:55.113 -> Build:Feb  7 2021
07:20:55.113 -> rst:0x13 (GLITCH_RTC_RST),boot:0xd (SPI_FAST_FLASH_BOOT)

`

Jason2866, 2021-08-04:
[Tasmota](https://tasmota.github.io/docs/) does work without wifi problems.
You can find a precompiled build for the ESP32-C3 to test [here](https://github.com/Jason2866/Tasmota-specials)

VojtechBartoska, 2022-03-30:
Hello, I'm closing this issue as it's outdated and seems it was solved already. Thanks for your contribution, if needed, you can reopen the issue.

closed 2022-03-30 by VojtechBartoska