espressif/arduino-esp32 #5495: Does the Arduino-ESP32 C3 support BLE Mesh?

lybgo, 2021-08-04:
Are there any examples if supported?
If not, are you ready to support it in the future?

me-no-dev, 2021-08-10:
if it's supported in ESP-IDF, then it is also supported here. Maybe ask there?

me-no-dev, 2021-08-11:
then it is also supported here :) Do you have an example in ESP-IDF that you want to try in Arduino?

chegewara, 2021-08-11:
> then it is also supported here :) Do you have an example in ESP-IDF that you want to try in Arduino?

Well, it is not that easy. BLE mesh is supported only when it is enabled in menuconfig explicitly.

lybgo, 2021-08-12:
> then it is also supported here :) Do you have an example in ESP-IDF that you want to try in Arduino?

I don't know how to use the ESP-IDF API in Arduino, can you give an example?

VojtechBartoska, 2022-03-30:
Hello, I'm closing this one as there is no traction. If needed, you can reopen it.

closed 2022-03-30 by VojtechBartoska