espressif/arduino-esp32 #5497: Test between 2.0.0-RC1 and 1.0.6 - Issue it Interrupt Functions

ale-trevizoli, 2021-08-04:
Hi.
Two important things happen when migrate same code compiled with 1.0.6 and 2.0.0-RC1.
One is Memory. Memory will be lower with 2.0.0-RC1.
Second very important: I'm using an Wiegand Library adaptated from Arduino to ESP32 (Yet another Wiegand Library from Paulo-Raca. I have to rewrite some functions using IRAM_ATTR because I have interrupts firing while read or write on flash.
But it's not the big problem. 
But on 1.0.6 It can works with no problem, on 2.0.0-RC1, I don't know if have problem with trigger interrupts, but from 10 tries of read an RFID card on simple wiegand reader, I have 9 with bit error reading and 1 I can receive complete number.
It's happens only if I'm reading and writing flash (FATFS) and version 2.0.0-RC1 is used, because I have a process that send a lot of information to a TXT file write in FLASH using POST HTTP.
on 1.0.6 same issue not happens.

I can't put code here, because it's very big and complex, but I review a lot of times, to check for bugs on code, and can't found.
The most important is that it's happen only in 2.0.0-RC1.
I'd like to know what changed in this version, that can produce this issue. I'm guessing is something with Interrupts routines and Flash routines.

Thanks a lot for all, for giving time and know-how to do this big framework as open source. To millions of developers can use it and create new products.
Sorry for my poor english. I'm Brazilian.

Jason2866, 2021-08-04:
No offense. How should a issue be found without code that demonstrate the problem or
a exact description what changes are made that are problematic.

SuGlider, 2021-08-11:
Please check #5474 regarding heap free memory.
We need more detailed information about the issue.

Please fill the issue template.

closed 2021-12-14 by VojtechBartoska