espressif/arduino-esp32 #5499: esp_hid_gap.h

mjs513, 2021-08-04:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Trying to compile and esp_hid_host example from the espressif_ide.

Describe your system( Hardware, computer, O/S, core version, environment).
Running on a Windows 10x64 pc using a ESP32 WROOM module.    Compiled using 2.0.0rc1.

Describe what is failing.
```ESP32_HID:28:10: fatal error: esp_hid_gap.h: No such file or directory```

Show the shortest possible code that will duplicate the error.
NA

Show the EXACT error message(it doesn't work is not enough).
```ESP32_HID:28:10: fatal error: esp_hid_gap.h: No such file or directory
 #include "esp_hid_gap.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1
esp_hid_gap.h: No such file or directory
```

lbernstone, 2021-08-05:
Did you include the esp_hid_gap.h (& .c) from the [example](https://github.com/espressif/esp-idf/tree/release/v4.2/examples/bluetooth/esp_hid_host/main) in your sketch folder?

mjs513, 2021-08-05:
I did now - how embarrassing - I should know better

closed 2021-08-05 by mjs513