espressif/arduino-esp32 #5500: PC connects with ESP32 SOFTAP with old SOFTAP SSID & Password Stored in PC

Electro-boy, 2021-08-05:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation version:			?1.0.6?
IDE name:						?Arduino IDE?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					115200
Computer OS:                                        ?Windows 8.1? 

#######PROBLEM####

I was using WIFI Soft AP in ESP32 Dev Board With the following parameters

**ssid : MYESP
password: ESP1234**

Connected PC with the SOFT AP credentials above.
It was working fine.

For another project I was trying to use the Same ESP32 DEV BOARD and uploaded the sketch with different SSID/password

**SSID: WIFIRELAY
Password: RELAYPASS**

now my PC wifi shows the OLD name when scanning "MYESP"
At beginning, I thought it wont connect ; but tried.

PC connected to the ESP32 - with old SSID/password. 

But the sketch contains New SSID/PASSWORD.

I Tried removing the stored network in PC and restarted the ESP32. But still have the same old SSID as ( MYESP)

ANY SOLUTIONS ??

### Sketch: 
```cpp

#include "WiFi.h"
#include <WiFiClient.h>
#include <WebServer.h>

const char * ssid = "WIFIRELAY";
const char * password = "RELAYPASS";

WebServer server(80);

void handleRoot() {
  
  server.send(200, "text/plain", "hello from esp32!");

}

void handleNotFound() {
  
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  
}

void setup() {

  Serial.begin(115200);
  
  WiFi.mode(WIFI_OFF);
  delay(1000);
  Serial.print("Setting AP (Access Point)…");
  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid, password);

  IPAddress IP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(IP);

server.on("/", handleRoot);
server.onNotFound(handleNotFound);

server.begin();
Serial.println("HTTP server started");

}

void loop() {

server.handleClient();
Serial.print("Stations connected: ");
Serial.println(WiFi.softAPgetStationNum());
delay(1000);
}
```

lbernstone, 2021-08-05:
Try [erasing the flash](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region).  You can find where esptool.exe is on your computer with a search.

VojtechBartoska, 2022-03-30:
Hello @Electro-boy, is this still valid?

If yes, please give it a try on latest development version 2.0.3-RC1. How to use this version is explained [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html).

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska