espressif/arduino-esp32 #5502: ESP32-C3 cant read all analog values woth analogRead()

kramzarales, 2021-08-05:
Hello,

I am currently programing an ESP32 C3 devboard which has multiple ADC channels and i would like to read data from 3 different sensors.
The problem is that whenever i use analogRead() with any port it always only shows the measurment on GPIO1.
I have tried it on an ESP32C3-WRoom-02 and also on the ESP32-C3 devmodule 02 with the same result.

I simply connected an external power supply to the pins and when connected to GPIO0 it reads the correct value, but on any other pin it only shows an analog value of 700-900 (i.e. floating pin)
Someone on ESP forum has also reported the same issue

### Hardware:
Board:							ESP32C3 Dev Module
Core Installation version:			1.0.0
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
When trying to read the analog value of any pin with analogRead(pin); it always reads only the value on GPIO0 (ADC1_CHANNAL0). I can can get any other analog-enabled pin to read the value

### Sketch:  
//Just a super basic analog read sketch
int sensorPin = 3;  
int sensorValue = 0;  

void setup() {
  // declare the sensorPin as an input doesnt change anything:

   Serial.begin(115200);
}

void loop() {
  // read the value from the sensor:
sensorValue = analogRead(sensorPin);
  Serial.println(sensorValue);
  delay(100);
}

### Debug Messages:
it compiles and uploads perffectly

wllis, 2021-09-19:
I got the same problem too !! so what should I do to solve this issue?

RudyFiero, 2021-09-19:
Same here. I tried to follow the code to see if it was a result of the IDF or if it was something in the Arduino code. I wasn't successful. I wanted to try just the IDF alone to check but have not yet found the time to do that.

kramzarales, 2021-09-20:
> I got the same problem too !! so what should I do to solve this issue?

I later figured out a solution. You can use the #include <driver/adc.h> library (dont need to download it) and then use the folowing example code:

setup:
adc1_config_width(ADC_WIDTH_BIT_12);
adc1_config_channel_atten(ADC1_CHANNEL_3,ADC_ATTEN_DB_11); //set channel atenuation to 11db (optional)
 ledcSetup(0, 1000, 8); //channel1, 1000Hz frequency, 8 bit resolution
 ledcAttachPin(PWMPin, 1);  //map any ADC suported pin to "channel 1"

loop:
 ledcWrite(1, duty cycle);  // set the duty cycle of "channel 1" (from 0 = 0% to about 1024 =100%)

wllis, 2021-09-20:
> I later figured out a solution. You can use the #include <driver/adc.h> library (dont need to download it) and then use the folowing example code:
>
> setup:
> adc1_config_width(ADC_WIDTH_BIT_12); [...]

Thanks !  I'll try it 😘

s-light, 2021-11-08:
i have found [this post at the arduino forum](https://forum.arduino.cc/t/problem-with-analogread-on-esp32-c3/892651/14)
>Theory
>I believe the problem is that the hardware abstraction layer define in the "esp32-hal-gpio.c" file never defines the pins for the C3 to begin with. ..... There are 2 #defines, one for the ESP32, and one for the ESP32S2, but never one for the ESP32C3. This holds true for both the esp32_gpioMux and the esp32_adc2gpio. I believe, if these arrays are filled out for the C3, this will start working?

> The Conclusion???
> I really looks like the C3 just doesn't have it's GPIO/ADC pins mapped. The first 22 GPIOs point to Analog channel 0. 
> And this A/D channel does indeed work as expected.

so the missing peaces are some defines in
[cores/esp32/esp32-hal-gpio.c#L49](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-gpio.c#L49)

something like
```
#elif CONFIG_IDF_TARGET_ESP32C3
const int8_t esp32_adc2gpio[5] = {1, 2, 3, 4, 5};
// GPIO5 == ADC2_0 so we do not want this one?!
```
and
```
#elif CONFIG_IDF_TARGET_ESP32C3
    {0x??,  0, -1, -1},
...
    {0, -1, -1, -1}
```
@me-no-dev could you add these? i have currently no idea how `esp32_gpioMux` works / what the values mean.....
( i mention you as you added initial c3 support - so i think you have read into the documentation already ;-)

s-light, 2021-11-08:
as my comment suggested..
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/adc.html#adc-limitations
ADC2 is used by Wifi - so it is only available if no wifi is used..

Jason2866, 2021-11-09:
@s-light ADC1 is fully useable. That ADC2 is conflicting with wifi is the same as for the ESP32

s-light, 2021-11-10:
i know. my point was:
do we want to include GPIO5 (ADC2_0) as usable adc channel?!
or do we leave it out in the first place as it conflicts if wifi is used...

i personally think it should be included.
(i did not check how the adc1 / adc2 differentiation is implemented in the other ports - and how much extra work this would be..)

NiklasVoigt, 2021-11-22:
Ran today into the same error with unusuable GPIO5 (ADC2_0) after manufacturing my pcbs...

mehulsharma21, 2021-12-02:
> i have found [this post at the arduino forum](https://forum.arduino.cc/t/problem-with-analogread-on-esp32-c3/892651/14)
>
>
>
> so the missing peaces are some defines in [...]

@s-light 
Hi, Please copy the below table and try.

#if CONFIG_IDF_TARGET_ESP32C3
    {0x04, -1,  0, -1},     // ADC1_CH0
    {0x08, -1,  1, -1},     // ADC1_CH1
    {0x0c, -1,  2, -1},     // ADC1_CH2 | FSPIQ
    {0x10, -1,  3, -1},     // ADC1_CH3
    {0x14, -1,  4, -1},     // MTMS | ADC1_CH4 | FSPIHD
    {0x18, -1,  5, -1},     // MTDI | ADC2_CH0 | FSPIWP
    {0x1c, -1, -1, -1},     // MTCK | FSPICLK
    {0x20, -1, -1, -1},     // MTDO | FSPID
    {0x24, -1, -1, -1},     //
    {0x28, -1, -1, -1},     //
    {0x2c, -1, -1, -1},     // FSPICSO
    {0x30, -1, -1, -1},     //
    {0x34, -1, -1, -1},     // SPIHD
    {0x38, -1, -1, -1},     // SPIWP
    {0x3c, -1, -1, -1},     // SPICSO
    {0x40, -1, -1, -1},     // SPICLK
    {0x44, -1, -1, -1},     // SPID
    {0x48, -1, -1, -1},     // SPIQ
    {0x4c, -1, -1, -1},     // USB-
    {0x50, -1, -1, -1},     // USB+
    {0x54, -1, -1, -1},     // U0RXD
    {0x58, -1, -1, -1},     // U0TXD

Jason2866, 2021-12-02:
With [this](https://github.com/tasmota/arduino-esp32/pull/31) changes it does now work. Thx @mehulsharma21

misery, 2021-12-28:
Could something like this be added in next release?
```
#elif CONFIG_IDF_TARGET_ESP32C3
const int8_t esp32_adc2gpio[5] = {1, 2, 3, 4, 5};
```

I get this on WLED because ```analogInputToDigitalPin``` is used which will point in ```variants/esp32c3/pins_arduino.h``` to ```esp32_adc2gpio```.
```
Linking .pio/build/esp32c3/firmware.elf
/home/andre/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c3/src/blynk.cpp.o: in function `BlynkApi<BlynkProtocol<BlynkArduinoClientGen<Client> > >::processCmd(void const*, unsigned int)':
/home/andre/git/WLED/wled00/src/dependencies/blynk/Blynk/BlynkApiArduino.h:93: undefined reference to `esp32_adc2gpio'
/home/andre/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/andre/git/WLED/wled00/src/dependencies/blynk/Blynk/BlynkApiArduino.h:93: undefined reference to `esp32_adc2gpio'
/home/andre/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/andre/git/WLED/wled00/src/dependencies/blynk/Blynk/BlynkApiArduino.h:95: undefined reference to `esp32_adc2gpio'
/home/andre/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/andre/git/WLED/wled00/src/dependencies/blynk/Blynk/BlynkApiArduino.h:101: undefined reference to `esp32_adc2gpio'
/home/andre/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c3/src/button.cpp.o: in function `isButtonPressed(unsigned char)':
/home/andre/git/WLED/wled00/button.cpp:89: undefined reference to `touchRead'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32c3/firmware.elf] Error 1
```

Bastlirovo, 2021-12-28:
@misery There was update to 2.0.2 5 days ago with some C3 updates. I didn't had time to check it in details, see yourself https://github.com/espressif/arduino-esp32/releases/tag/2.0.2

misery, 2021-12-28:
Thanks, I already tried 2.0.2 without success. I added the variable in my fork. It compiles fine. But not tested.

Seenov, 2022-02-13:
The implementation of the ADC for the ESP32-C3  for the Arduino platform must be fixed to  provide the same level of performance as the esp-idf implementation.
Hardware:
Board: ESP32C3  rev 3 Dev Module 
Core Installation version: 2.0.0
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Linux

What works in the IDF version can be seen with VScode and the /examples/peripherals/adc/single_read example code. Both ADC1 and ADC2  work. The returned values are with 1% of the actual values: the following table gives the results:

Channels--Voltmeter---Arduino---Arduinodiff---diff%---VScode---VsCodediff---VsCode%
ADc1-0---------0.47--------0.309--------0.161--------34%------0.469---------0.001-------------0%
--ch1------------0.9403-----0.615--------0.3253------35%------0.934	--------0.0063-----------1%
----2--------------1.412-----0.934--------0.478---------34%------1.417	--------0.005-------------0%
----3	-------------1.8812-----1.25---------0.6312-------34%------1.902--------0.0208-----------1%
----4	-------------2.355------1.573--------0.782---------33%------2.382--------0.027------------1%
ADC2ch0-----1.412---------n/a--------------------------------------1.393---------0.019------------1%
Notes: -The Arduino code averages 16 samples and the WiFi is running.  The VsCode example was modified to sample all channels, WiFi not active, one sample per channel.

Comparing the Arduino  esp32-hal-adc.c code  with the esp-idf single-read example shows that the calibration  functionality for the esp32-C3 is not implemented in the Arduino code. This needs to be fixed.

The issues of ADC2

-Adding the pin definitions for the ESP32-c3 as suggested in this post does not seem to change anything. 
-My Arduino code causes continuous reboots when I try to activate ADC 2 with or without WiFi
code: 
`adcAttachPin(5);    
 analogSetPinAttenuation(5, ADC_11db);
Serial.println(analogReadMilliVolts(5));
`
Debug message:
E (5285) ADC: adc1_config_channel_atten(475) :ADCADC_NUM_1 channel error
E (5286) ADC: adc1_config_channel_atten(475) :ADCADC_NUM_1 channel error
E (7025) ADC: adc1_config_channel_atten(475) :ADCADC_NUM_1 channel error

assert failed: esp_efuse_rtc_calib_get_init_code esp_efuse_rtc_calib.c:22 (atten < 4)
Core  0 register dump:
MEPC    : 0x40381b4c  RA      : 0x40386bc2  SP      : 0x3fc9c560  GP      : 0x3fc8da00  
TP      : 0x3fc85fd0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x3fc9c6d5  S1      : 0x00000072  A0      : 0x3fc9c5b8  A1      : 0x3fc8f425  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc96000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fc9c5b8  S3      : 0x00000001  
S4      : 0x3fc9c5b8  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

The first issue is getting the code to work. Eliminating ADc2 should not be considered. My project is a good example of the use of ADC2. It is a WiFi 6 channel ADC with an optional battery. In battery mode the ESP32-C3 must be put in deep sleep to conserve power, turning off the WiFi. After turning off the WiFi, the battery voltage can be measured with ADC2.

The second issue is the conflict with the WiFi and how that impacts using ADC2.
Going back to the single_read example:
code exert:
```
while (1) {
        adc_raw[0][0] = adc1_get_raw(ADC1_EXAMPLE_CHAN0);
        ESP_LOGI(TAG_CH[0][0], "raw  data: %d", adc_raw[0][0]);
        if (cali_enable) {
            voltage = esp_adc_cal_raw_to_voltage(adc_raw[0][0], &adc1_chars);
            ESP_LOGI(TAG_CH[0][0], "cali data: %d mV", voltage);
        }
        vTaskDelay(pdMS_TO_TICKS(1000));
        do {
            ret = adc2_get_raw(ADC2_EXAMPLE_CHAN0, ADC_WIDTH_BIT_DEFAULT, &adc_raw[1][0]);
        } while (ret == ESP_ERR_INVALID_STATE);
        ESP_ERROR_CHECK(ret);

        ESP_LOGI(TAG_CH[1][0], "raw  data: %d", adc_raw[1][0]);
        if (cali_enable) {
            voltage = esp_adc_cal_raw_to_voltage(adc_raw[1][0], &adc2_chars);
            ESP_LOGI(TAG_CH[1][0], "cali data: %d mV", voltage);
        }
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
```
There is a difference between the implementations of ADC1 and 2.  ADC1 can be read any time by ADC1_get_raw and the results do no need to be qualified.
ADC2  adc2_get_raw returns one of four possible values:  ESP_OK, ESP_ERR_INVALID_STATE, ESP_ERR_TIMEOUT and ESP_ERR_INVALID_ARG.
Only ESP_OK corresponds to a valid reading. (the while statement in the code is actually wrong, it should be while(ret!= ESP_OK)).
This means that the Arduino implementation of ADC2 should expose this qualification of the validity of the ADC2 results.

me-no-dev, 2022-02-13:
it says you are using version 2.0.0? Have you tried with 2.0.2 or master? We have changed the driver to use ESP-IDF

Seenov, 2022-02-14:
How do you know which version you have?
If I look in the platform.txt file, in my working directory it is 2.0.0, if I download the asset file esp32-2.02  it is version=        (blank), if I look at the one on git master branch it is 2.0.0.
I will try the esp32-2.0.2 and post the results.
Thanks

Seenov, 2022-02-14:
I just tried 2.0.2.  No improvement, values have not changed. I believe these 2 statements generate the error listed below:
`adcAttachPin(5);              //confirm how to do this for ADC2 ch0
   analogSetPinAttenuation(5, ADC_11db);  // confirm how to do this for ADC2 ch0
         output on serial monit: E (21051) ADC: adc1_config_channel_atten(475) :ADCADC_NUM_1 channel error
`
If I try to read:
`Serial.println(analogReadMilliVolts(5));`
I get constant reboots.
Full code:
`if(millis() >= time_2 + INTERVAL_2){
        time_2 +=INTERVAL_2;
        
        WiFi.disconnect(); // ensure a clean start
        adcAttachPin(5);              //confirm how to do this for ADC2 ch0
        analogSetPinAttenuation(5, ADC_11db);  // confirm how to do this for ADC2 ch0
        Serial.println(analogReadMilliVolts(5));
        for (int i = 0; i < 32; ++i)
        {
        esid += char(EEPROM.read(i));
        }
        for (int i = 32; i < 96; ++i)
        {
        epass += char(EEPROM.read(i));
        }
        WiFi.begin(esid.c_str(), epass.c_str());
        WiFi.mode(WIFI_STA);
        
     }
`

mdjavedakhtar, 2022-02-26:
Anyone got this working??

Seenov, 2022-02-27:
Is this the refactoring you are talking about?
ESP32 Arduino 2.0.2 based on ESP-IDF 4.4-beta1 
add GPIO defines for C3 by @NiklasVoigt in https://github.com/espressif/arduino-esp32/pull/6005
I have copied the esp32-2.0.2.zip asset to my Arduino install and tested the ADC. ADC1 values still off by 30% and ADC2  generated errors and reboots.
code:
`adcAttachPin(5);              //confirm how to do this for ADC2 ch0
        analogSetPinAttenuation(5, ADC_11db);  // confirm how to do this for ADC2 ch0
        for (char samples1=0; samples1<NumOfSamples_per_Avrg[0]; samples1++)
        {
            chan_summ[5]+=analogReadMilliVolts(5);
        }`
Generated errors (plus reboot):
`E (5323) ADC: adc1_config_channel_atten(475) :ADCADC_NUM_1 channel error
E (5330) ADC: adc1_config_channel_atten(475) :ADCADC_NUM_1 channel error
E (5338) ADC: adc1_config_channel_atten(475) :ADCADC_NUM_1 channel error
assert failed: esp_efuse_rtc_calib_get_init_code esp_efuse_rtc_calib.c:22 (atten < 4)

`
If you read the code of esp32-hal-adc.c it is evident there is no support for ESP32C3  ADC1/2 calibration and no support for ESP32-C3 ADC2.

regards

Jason2866, 2022-02-27:
No, this one https://github.com/espressif/arduino-esp32/pull/6259
which completly changes gpio handling

Jason2866, 2022-02-28:
> The ADC works fine on VS code so a third development platform is not necessary.

Does it work for you or not? VS code uses Platformio. I do not get what you mean with 3rd Platform?

Seenov, 2022-02-28:
I am on Linux and used PlarformIO with Atom for another project, I forgot it also works with VS code.
On Arduino no calibration, ADC2 does not work. I will release my board with those limitations adding a guide to show customers how to calibrate the ADC1. My 6 channel ADC is now a 5 channel ADC.
My board has a battery option. To conserve power while on battery the ESP32C3 was going to be put into deap sleep most of the time, at wake up it was going to read the battery voltage before turning on the wifi. So ADC2 is very usefull even with the wifi conflict.
All the best

Seenov, 2022-03-04:
I am disapointed having to resolve these issues myself.  I have a temporary workaround to get ADC1 calibration working. I have found this library  and modified it for the ESP32C3: https://github.com/madhephaestus/ESP32AnalogRead.
The results are the same as the VsCode implementation with 1% of measured values. The only problem left is ADC2 and it looks like the attachpin function may be the culprit. When I atempt to attach pin5, I get errors or reboots.  The status should be changed and a solution found so ADC2 can work properly.

Seenov, 2022-03-04:
There is nothing like being persistent. As I worked on this issue I finally realised that I can simply copy Vscode from ESP-IDF into Arduino and it will compile. 
The results: Calibration works on all channels even ADC2_chan0
ADC2 work with wifi! Yes , my example code has wifi turned on and ADC2_chan_0 give accurate results.
I have suspected wifi uses ADC2 for a very short time and if my request to do a conversion fall outside of the wifi priority, the conversion will take place. More testing will be required to ensure there is no adverse effect on the wifi.
This is my code example:
https://seenov.com/2022/03/04/arduino-esp32c3-6-channel-adc-with-calibration/
Hve fun!

VojtechBartoska, 2022-04-13:
Thanks for testing, I'm closing the issue.

closed 2022-04-13 by VojtechBartoska