espressif/arduino-esp32 #5503: Very inaccurate analogRead on GPIO36 with latest core installation

proboscide99, 2021-08-05:
### Hardware:
ESP32-WROOM-32E
Programmed with board type 'DOIT ESP32 DEVKIT V1'
using arduino ide V1.8.13 and 1.8.15 under Windows and Ubuntu (all combination, same result)

### Description:
Hello, 

I have a project running on ESP32-WROOM-32E in which I use GPIO36 (ADC1_CHANNEL_0) as analog input and everything worked fine until I updated the esp32 core to latest version (4.4). I was previously running the one-year old V3.3. The Arduino IDE version is 1.8.13 and 1.8.15 (same result).

With the new core, the value returned by 'analogRead(36)' is rather non linear (I can still read 0 or 4095 but, for example, with 1/2Vdd input I read 4095 instead of ~2047). I get ~3000 with a voltage near to zero on the pin.

I read in the esp32 documentation that this pin is shared with an internal hall sensor, but for some reason, this was not an issue with the older core.

I compared the adc-related source code and found that the hallRead function in the older core used to initialize and deinitialize some internal registers, while the newer one does not. I then tried to do the same (executing the de-init code in the sketch) in the hope to get the hall sensor disabled but got no result.

Output with older core (v3.3.1-61-g367c3c09c):

20:50:39.589 -> AD Val = 34
20:50:40.090 -> AD Val = 35
20:50:40.591 -> AD Val = 35
20:50:41.091 -> AD Val = 34
20:50:41.591 -> AD Val = 35

Output with latest core (v4.4-dev-2313-gc69f0ec32), same voltage on GPIO:

20:53:31.340 -> AD Val = 145
20:53:31.806 -> AD Val = 160
20:53:32.306 -> AD Val = 162
20:53:32.840 -> AD Val = 154
20:53:33.307 -> AD Val = 137
20:53:33.841 -> AD Val = 135

Is there a way to get back the analog GPIO36 capability as it was on the V3.3 core?

Thanks a lot and Best Regards!

### Sketch:

```
  void setup() {
  Serial.begin(115200);

  pinMode(36, ANALOG);
}

void loop() {
  delay(500);
  Serial.print("AD Val = ");
  Serial.println(analogRead(36));
}
```

felmue, 2021-08-06:
Hello @proboscide99 

I think what has changed is the default attenuation. It used to be 11dB but it looks like it is now 0dB.

Try adding this line in your setup code:
```
analogSetAttenuation(ADC_11db);
```
Thanks
Felix

proboscide99, 2021-08-06:
Thanks a lot! The 'analogSetAttenuation' didn't solve the problem, but then I searched for a pin-specific setting (because GPIO36 was the only pin affected by the attenuation shift) and that solved the problem!!!

analogSetPinAttenuation(AN_JOYSTICK_PIN, ADC_11db); 

For a reason I can't explain, the "global" function you suggested didn't work, while IMHO it should have.

Thanks again
Alessandro

felmue, 2021-08-06:
Hello @proboscide99

interesting - thank you for sharing. I am glad to hear you got it working.

Thanks
Felix

marcboon, 2021-08-10:
> Thanks a lot! The 'analogSetAttenuation' didn't solve the problem, but then I searched for a pin-specific setting (because GPIO36 was the only pin affected by the attenuation shift) and that solved [...]

In 2.0.0-rc1 The function analogSetAttenuation() only sets a local (static) variable to that value. The value is used in the calibration functions, but not for a simple analogRead(). Use the analogSetPinAttenuation() function for each pin you intend to use. You only need to set it once (for example in setup()).
The default attenuation for all pins is 0dB, which gives a full-scale output at about 800mV input voltage. With 11dB attenuation the full-scale range is about 2.55V. 

I'm using a ESPS2, where the ADC width is 13 bit by default, so full scale gives 8191 as a result. I didn't test other ESP32 versions.

closed 2021-08-11 by me-no-dev