espressif/arduino-esp32 #5510: RTC_SLOW_MEM[128:131] corrupted in 2.0.0-rc1

victor-chew, 2021-08-08:
### Hardware:
Board:							ESP32 D1 Mini
Core Installation version:			2.0.0-rc1
IDE name:						PlatformIO
Flash Frequency:					80MHz
PSRAM enabled:                                     Yes
Upload Speed:					912600
Computer OS:                                         Windows 10

### Description:
I find that RTC_SLOW_MEM[128 - 131] is corrupted when waking up from deep sleep. This only happens with 2.0.0-rc1. It does not happen with 1.6.6.

The sample code loads a simple ULP program into memory (basically just 4 instructions - wait for wake flag to become available, wake and halt). It then takes a snapshot of the entire RTC_SLOW_MEM and saves it to flash, then goes into deep sleep. 

When the ULP program wakes up the SoC, the snapshot is read from flash and compared to the RTC_SLOW_MEM. The values at DWORD[128 - 131] are consistently corrupted when compiled with 2.0.0-rc1. With 1.6.6, no corruption occurs.

### Sketch:
I have attached the platformio.ini and cpp file that can be used to replicate this issue.
(https://github.com/espressif/arduino-esp32/files/6950693/test_ulp.zip)

igrr, 2021-08-09:
@victor-chew RTC slow memory may be used by the linker to allocate variables. It is not guaranteed that any particular location is free to use, unless you have reserved it by allocating a variable in RTC slow memory. It means this might accidentally work, but the linker is also free to place some variable into the unused space. You need to use `RTC_SLOW_ATTR` attribute to place a static variable into RTC slow memory, e.g. `uint32_t RTC_SLOW_ATTR myvar;`. Then you can access `myvar` in your program.

victor-chew, 2021-08-09:
@igrr: Thanks for your response. The reason why I came across this problem is because I am working on a ULP program that is > 128 words. So I took the wordaround given here:

https://www.esp32.com/viewtopic.php?t=7023

and patched the ulp_process_macros_and_load() function. This was working under 1.6.6.

However, with 2.0.0-rc1, RTC_SLOW_MEM[128:131] would get trashed, as I have described. I am guessing from your answer that the linker assumes 0-127 is for the ULP code, and it can use everything after that as it chooses.

So is there an official way to load a ULP program that is larger than 128 words, and tell the linker to only use the space after that for its variables?

Thanks!

igrr, 2021-08-09:
I'm afraid that the "official" way is to [use Arduino-esp32 as an ESP-IDF component](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html), which lets you modify `CONFIG_ULP_COPROC_RESERVE_MEM` in menuconfig.

If you think the default value of CONFIG_ULP_COPROC_RESERVE_MEM in Arduino-esp32 should be changed, you might convince the maintainers of this repo to do this.

It might also be possible to introduce some build-time approach for changing this value in the linker script, but that would mean adding another menu in Arduino IDE, and there already are quite a few menus...

VojtechBartoska, 2022-03-30:
Hello, as there is no traction. I'm closing this issue. If need, you can reopen it @victor-chew. Thanks for your contribution.