espressif/arduino-esp32 #5512: uartAvailable() has race condition

rob-deutsch, 2021-08-09:
I am experiencing race conditions with `HardwareSerial.available()` and `HardwareSerial.read()` which seem to be due to the underlying HAL `uartAvailable()` function.

In particular, the following line seems to have a race conditions:

https://github.com/espressif/arduino-esp32/blob/5bb8177aa1feec665938be97607b526a4f0ec0bb/cores/esp32/esp32-hal-uart.c#L344

In my code I call `HardwareSerial.available()` which returns >0, but then when I call `HardwareSerial.read()` I get -1 indicating that there is no data to read.

I think that `uartAvailable()` should turn off interrupts and call `UART_MUTEX_LOCK()` in order to avoid the race condition.

SuGlider, 2021-08-11:
You should have filled the issue template fully. 
It is not clear what version of ESP32-Arduino you are running. 
Please provide more information.

rob-deutsch, 2021-08-12:
Thank you @SuGlider, I have updated the top post.

My apologies I didn't use the template. The template didn't come up when I used Github's "reference in new issue" link.

SuGlider, 2021-08-12:
Thanks for the extra information @rob-deutsch.

I could not reproduce such issue. I tested it using Core 1.0.6 and 2.0.0RC1.
Please run the sketch below to find out the Arduino Core and IDF version you are currently running, and then report the output back to us.

''' javascript
#include <Arduino.h>
#include "core_version.h"

void setup() {
    Serial.begin(115200);

    Serial.println("IDF Version:");
    Serial.println(ESP.getSdkVersion());

    Serial.println("Arduino Core Version:");
    Serial.println(ARDUINO_ESP32_RELEASE);
}

void loop() {
}
'''

rob-deutsch, 2021-08-13:
Thank you SuGlider, I have added that information to the original post.

I've also given a better demonstratino Sketch, with an added call to a function called DoOtherStuff(). This is a race situation, so the error doesn't always occur.

SuGlider, 2021-08-13:
@rob-deutsch, please notice that uxQueueMessagesWaiting() doesn't consume any data and reading the value of the Register status.rxfifo_cnt also doesn't consume any data from FIFO.

Maybe the racing condition issue is Application related and not Arduino Core related.

Anyway esp32-hal-uart.c has changed since version 1.0.6.
A newer version 2..0.0 is about to be relaeased.
At this point PlatformIO only supports 1.0.6.

SuGlider, 2021-08-13:
Hi @rob-deutsch. thank you for your time on this issue. 

I tried to reproduce the issue several times and it always went well (in both 1.0.6 and 2.0.0)
Could you please provide sample code to effectively reproduce this problem?

SuGlider, 2021-08-13:
Hi @rob-deutsch, thank you for pointing out this issue.

In fact, in case there is a task reading the  UART at the same time that another task  tries to get uartAvailable() on this same UART, a race condition may happen and the information returned could be outdated.

Currently the code of esp32-hal-uart.c is being refactored.
This issue will be solved in the core verion 2.

I'll post back here as soon as a correction is released.

SuGlider, 2021-08-14:
esp32-hal-uart.c is being refactored. This issue will be solved in the process.

closed 2021-08-14 by SuGlider