espressif/arduino-esp32 #5514: FreeRTOS: incorrect  portSTACK_TYPE definition

dibalavs, 2021-08-09:
### Hardware:
|||||||
|:---|---|---|---|---|---|
|Board|ESP32 Dev Module||
|Version/Date|framework-arduinoespressif32 3.10006.210326 (1.0.6)|
|IDE name|VSCode + Platform.io|
|Flash Frequency|80Mhz|
|PSRAM enabled|yes|
|Upload Speed|115200|
|Computer OS|Windows 10|

### Description:
Describe your problem here

When use xTaskCreateStatic()  you need to declare stack as separate static variable:

```
StackType_t xStack[ STACK_SIZE ];
```
due to official documentation:  https://www.freertos.org/xTaskCreateStatic.html
(also see an example on this link)
STACK_SIZE means size in **words** not bytes, but in  arduino-esp32 
StackType_t is defined as (portmacro.h):

```
#define portSTACK_TYPE	uint8_t
typedef portSTACK_TYPE			StackType_t;

```

The issue is ising uint8_t, not uint32_t

as a result I have a crash even on simple task which do only log message:

### Sketch:  
```cpp

#define UART_STACK_WORDS 400
#define UART_LOOP_DELAY_MS 20

StaticTask_t uart_task_buf;
StackType_t uart_task_stack[UART_STACK_WORDS];

void task_uart(void *args)
{
    (void)args;
    log_i("Start UART task");

    for (;;) {
        delay(UART_LOOP_DELAY_MS);
    }
}

void StartWorkerTasks()
{
    xTaskCreateStatic(task_uart,
        "TaskUart",
        UART_STACK_WORDS,
        NULL,
        tskIDLE_PRIORITY,
        uart_task_stack,
        &uart_task_buf);
}

```

### Debug Messages:
```
***ERROR*** A stack overflow in task TaskUart has been detected.
abort() was called at PC 0x40085134 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40084ea8:0x3ffc0020 0x4008511d:0x3ffc0040 0x40085134:0x3ffc0060 0x40086a23:0x3ffc0080 0x400885ec:0x3ffc00a0 0x400885a2:0x3ffc1400 0x400f3fc5:0x00000000
```

I don't see even "Start UART Task" in log message

igrr, 2021-08-09:
@dibalavs I think `StackType_t uart_task_stack[UART_STACK_WORDS];` is a correct way to define the stack size. However 400 bytes is too small size for a task on ESP32. Minimal stack size is around 800 bytes, but if you are using printf family functions (log_i is using printf internally), you need significantly larger stack size, not less than 2000 bytes.

(Indeed for legacy reasons, ESP-IDF is using uint8_t as StackType_t. If you want to write code that will behave the same regardless of sizeof(StackType_t), you can use `STACK_SIZE_BYTES / sizeof(StackType_t)` when specifying stack sizes, whether with static or dynamic stack allocation.)

VojtechBartoska, 2022-03-30:
Hello, as there is no traction. I'm closing this issue. If need, you can reopen it @dibalavs. Keep in mind UART was refactored and it needs to be validated against v2.0.3-RC1. Thanks for your contribution.

closed 2022-03-30 by VojtechBartoska