espressif/arduino-esp32 #5515: [start_ssl_client():216]: (-30336) SSL - No CA Chain is set, but required to operate

rishadali, 2021-08-09:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I am trying to use this WiFiClientPSK.ino sketch to create a connection with my Spring Boot Rest API. I have configured the api to work on the SSL connection, when I try to make a connection from the ESP32, I get ' SSL - No CA Chain is set, but required to operate'. 
I have checked this with the self-signed certificate as well as the live secure api but getting same issue. 
Is there any other way I have to implement on the server side to accept the PSK connections?
Please can you help resolve the issue?

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <WiFiClientSecure.h>

#if 0
const char* ssid     = "SSID";     // your network SSID (name of wifi network)
const char* password = "password"; // your network password
#else
const char* ssid     = "SSID";     // your network SSID (name of wifi network)
const char* password = "password"; // your network password
#endif

//const char*  server = "secure.api.com";  // Server hostname
const IPAddress server = IPAddress(192, 168, 194, 71);  // Server IP address
const int    port = 8443; // server's port (8883 for MQTT)

const char*  pskIdent = "Client_identity"; // PSK identity (sometimes called key hint)
const char*  psKey = "1a2b3c4d"; // PSK Key (must be hex string without 0x)

WiFiClientSecure client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    // wait 1 second for re-trying
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  client.setPreSharedKey(pskIdent, psKey);

  Serial.println("\nStarting connection to server...");
  if (!client.connect(server, port)){
    //Serial.println("Connection failed!");
    Serial.println("Connection failed!");
    //Serial.println("SSL error was: %u", client.errorCode());
  }
  else {
    Serial.println("Connected to server!");
    // Make a HTTP request:
    client.println("GET /a/check HTTP/1.0");
    client.print("Host: ");
    client.println(server);
    client.println("Connection: close");
    client.println();

    while (client.connected()) {
      
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available()) {
      char c = client.read();
      Serial.write(c);
    }

    client.stop();
  }
}

void loop() {
  // do nothing
}
```

### Debug Messages:
```
13:32:23.282 -> Starting connection to server...
13:32:23.282 -> [V][WiFiClientSecure.cpp:146] connect(): start_ssl_client with PSK
13:32:23.282 -> [V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 280384
13:32:23.282 -> [V][ssl_client.cpp:65] start_ssl_client(): Starting socket
13:32:23.349 -> [V][ssl_client.cpp:104] start_ssl_client(): Seeding the random number generator
13:32:23.349 -> [V][ssl_client.cpp:113] start_ssl_client(): Setting up the SSL/TLS structure...
13:32:23.383 -> [V][ssl_client.cpp:141] start_ssl_client(): Setting up PSK
13:32:23.383 -> [V][ssl_client.cpp:197] start_ssl_client(): Setting hostname for TLS session...
13:32:23.383 -> [V][ssl_client.cpp:212] start_ssl_client(): Performing the SSL/TLS handshake...
13:32:23.587 -> [E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-30336) SSL - No CA Chain is set, but required to operate
13:32:23.587 -> [E][WiFiClientSecure.cpp:153] connect(): start_ssl_client: -30336
13:32:23.621 -> [V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
13:32:23.621 -> Connection failed!
```

VojtechBartoska, 2022-03-30:
Hello @rishadali , sorry for late reply. Are you able to test your issue on development version 2.0.3-RC1 to check if it is still valid?. You can take a look on [hDocs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release in Arduino IDE.

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska