espressif/arduino-esp32 #5516: ESP32 wakeup-Timer gets restart when wake up from deep sleep from external interrupt

Rbhagwat62, 2021-08-10:
**Hardware:** 
board:ESP32 dev kit v1
IDE: Arduino
ESP32 stack : 1.0.6

**Description:**
My requirement is to use ESP32 in deepsleep, wakeup it after every one hour send the recorded data of one hour to the server. Recorded data will get store when any external interrupts generate on multiple IOs. 

I have referred to the sample code from the examples. In the code I have enable both timer and external interrupt by calling both API one after other in the setup loop

esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);  // for timer based wakeup
esp_sleep_enable_ext1_wakeup(BUTTON_PIN_BITMASK,ESP_EXT1_WAKEUP_ANY_HIGH); //wakeup when any masked pin High 

ESP32 is getting wakeup by timer as well as by any external interrupt. Only Issue is whenever I receive any external interrupt timer start from zero again and ESP wakeup after time is over.

My expectation is when ever external interrupt received timer should not restart, such that I will get data between that one hour.

I tried enabling timer only once in the initialization when ESP not wake up from deep sleep. It then gets wake up from deep sleep only once.

How can I achieve my requirement?

igrr, 2021-08-13:
@Rbhagwat62 I think you will need to use `gettimeofday` function to get the current time and calculate the remaining time to sleep based on that. Unfortunately it's not possible to keep the wakeup time counting down after deep sleep wakeup, it needs to be set up each time before deep sleep.

Rbhagwat62, 2021-08-18:
@igrr That was my secondary plan to implement, Thanks of the update

closed 2021-08-18 by Rbhagwat62