espressif/arduino-esp32 #5519: Core 0 Panic when reassigning Serial2 GPIO pins

JDSenseright, 2021-08-10:
I'm currently trying to reassign Serial2 on an ESP32 custom development board between GPIO pins 14,22 and 2,32.  

I need to be able to do this multiple times in a sketch to talk to two channels of sensors via modbus.

The below sketch however crashes with a Core 0 panic after some number of changes.

```
#include <Arduino.h>

HardwareSerial SerialOne(0);
HardwareSerial SerialModem(1);
HardwareSerial SerialSensors(2);

void setup() {
  SerialOne.begin(115200, SERIAL_8N1, GPIO_NUM_3, GPIO_NUM_1);
  SerialModem.begin(115200, SERIAL_8N1, GPIO_NUM_16, GPIO_NUM_17);
  SerialSensors.begin(9600, SERIAL_8N1, GPIO_NUM_14, GPIO_NUM_22);
}

void loop() { 

  delay(2000);
  SerialOne.println("Changing serial Sensors to D2");
  SerialSensors.begin(115200, SERIAL_8N2, GPIO_NUM_2, GPIO_NUM_32);
  delay(2000);
  SerialOne.println("Changing serial Sensors to D1");
  SerialSensors.begin(115200, SERIAL_8N2, GPIO_NUM_14, GPIO_NUM_22);
}
```

### Debug Messages:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Changing serial Sensors to D1
Changing serial Sensors to D2
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e25fb  PS      : 0x00060b30  A0      : 0x80086128  A1      : 0x3ffb78b0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x3ffb796c  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x800e25fb  A9      : 0x3ffb7870  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffaffe8  A13     : 0x00000000
A14     : 0x3ffaff9c  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000b  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400e25fb:0x3ffb78b0 0x40086125:0x3ffb78d0

Rebooting...
```

me-no-dev, 2021-08-10:
You should have filled the issue template fully. It is not clear what version of ESP32-Arduino you are running and you have not decoded the backtrace. You should try this with 2.0.0-rc1, because it already contains a few fixes for Serial.begin issues

JDSenseright, 2021-08-11:
> You should have filled the issue template fully. It is not clear what version of ESP32-Arduino you are running and you have not decoded the backtrace. You should try this with 2.0.0-rc1, because it already contains a few fixes for Serial.begin issues

Sorry me-no-dev, i've updated with the required information and i'll add the decoded backtrace tonight when i'm back in front of the computer.

shawty, 2021-08-11:
I'm also seeing this with a very simple example:

[image]

My ESP 32 Panics and Reboots as soon as it hits the SerialTwo.begin line, I'm mapping my Serial lines to PINS 6 & 7 as per this diagram:

[image]

As the devkit 1/doit board is the one I have.

My ESP32 is just caught in a panic/reboot loop.  If I comment out the SerialTwo.begin line and reboot, everything works as expected with the normal Serial1 object.

chegewara, 2021-08-11:
@shawty Pins 6/7 are flash pins and as you can see you dont even have them on this devkit.

shawty, 2021-08-11:
@chegewara oh for flips sake, I just realised what I was doing (Ironically something I'm good at... 'being a muppet'), I counted the pin numbers up the side of the board starting from the lower right (Where the 3.3v is) when what I should have looked at are the "GPIO Numbers"

So 6/7 6th & 7th one up from the lower right...

I'll go give myself a good slap now...

If this had been a PICMicro I would have been fine, dunno why I brain flipped like that!

SuGlider, 2021-08-11:
I tested the sketch presented for this issue with core version 2.0.0 RC1. 
It worked fine for more than 10 minutes. No reset or panic.

Issue is related to Arduino Core v1.0.6. and it is solved on new version 2.0.0.

SuGlider, 2021-08-11:
> You should have filled the issue template fully. It is not clear what version of ESP32-Arduino you are running and you have not decoded the backtrace. You should try this with 2.0.0-rc1, because it already contains a few fixes for Serial.begin issues

Yes, it is solved  with 2.0.0-rc1.

PR #5549 (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF
## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.
commit b1d072df (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF (#5549)

## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.
closed 2021-08-23 by me-no-dev