espressif/arduino-esp32 #5520: assertion "tcp_receive: valid queue length" failed occurs at random intervals

philbowles, 2021-08-10:
### Hardware:

[image]

<B>Version</B> 1.0.6
<B>IDE name</B> Arduino IDE 1.8.15
<B>PSRAM enabled</B> no
<B>Computer OS</B> Windows 10

### Description:

Why is this happening apparently at random?

assertion "tcp_receive: valid queue length" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c", line 1197, function: tcp_receive

Same behaviour shown on Wemos LOLIN D32

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include<WiFi.h>
#include "esp_task_wdt.h"

extern "C" {
    #include "lwip/tcp.h"
};
#define MS_INTERVAL 500
#define BURST_SIZE 10
#define MSG_SIZE 30

struct tcp_pcb* pcb;
bool   cnx=false;

uint8_t msg[MSG_SIZE];

err_t _tcp_connected(void* arg, void* tpcb, err_t err){
  Serial.printf("Connected to echoserver\n");
  cnx=true;
  tcp_nagle_disable(pcb);
}

void setup(){
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin("XXXXXXXX", "XXXXXXXX");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }
  Serial.printf("\nIP: %s\n",WiFi.localIP().toString().c_str());
  pcb=tcp_new();
  IPAddress ip(192,168,1,20); // 192.168.1.20:8080 is an "echo server"
  ip_addr_t ipt;
  ip_addr_set_ip4_u32(&ipt, ip);
  err_t err = tcp_connect(pcb, &ipt, 8080,(tcp_connected_fn)&_tcp_connected);
}

uint32_t T;
int ticker=0;

void loop(){
  if(cnx && ((millis() - T) > MS_INTERVAL)){
    T=millis();
    if(++ticker > 59){
      Serial.printf("TICK\n",millis());
      ticker=0;
    }

    for(int i=0;i<BURST_SIZE;i++){
      if(MSG_SIZE < tcp_sndbuf(pcb)) tcp_write(pcb,(const uint8_t*) msg,MSG_SIZE,TCP_WRITE_FLAG_COPY);
      else esp_task_wdt_reset(); // spin until space available     
[... 159 lines cut ...]
        print("RX ",len(data))
        print(data)

        if(data):
            self.send(data)

class EchoServer(asyncore.dispatcher):

    def __init__(self, host, port):
        asyncore.dispatcher.__init__(self)
        self.create_socket(socket.AF_INET, socket.SOCK_STREAM)
        self.set_reuse_addr()
        self.bind((host, port))
        self.listen(5)

    def handle_accept(self):
        pair = self.accept()
        if pair is not None:
            sock, addr = pair
            print ('Incoming connection from %s' % repr(addr))
            handler = EchoHandler(sock)

server = EchoServer('0.0.0.0', 8080)
asyncore.loop()
```

Braehead, 2021-08-11:
This is a show stopper for the ESP32 and is an intolerable issue that needs resolving by Espressif in the ESP32 Core Blob (and possibly the ESP8266 and other similar core libraries) as soon as possible.  This is a fundamental WiFi failure that kills the use of the product (and related platforms) dead in its tracks.

It would be interesting to know how long this issue has existed in the core and its span across the product range to understand if the random WiFi fails that have existed and plagued the ESPs for many years is a generic issue, been getting worse or just an unlucky introduction.

It is now time that the core blob is made a little more public to allow peer review for some of the fundamental and security issues of the ESP family core libraries,

philbowles, 2021-08-11:
@Braehead As we speak, I am recompiling for the ESP8266 - the code base is 98% identical (with some minor hardware-specific abstractions of course) but the LwIP code is identical. If it stays alive on the 8266 it might help narrow the problem to an ESP32+LwIP issue: If it fails in the same way then the finger points at LwIP alone...

philbowles, 2021-08-11:
For what it is worth the equivalent code in ESP8266 (identical in LwIP logic) has been running happily for over 2.5 hours now. Also I have not experienced any failures in LwIP in this part of the code on ESP8266 - which is identical in both my implementations. My money is therefore currently on the specific ESP32 implementation of LwIP and a lot less likely to be LwIP itself.

```cpp
#define LWIP_INTERNAL
#include<ESP8266WiFi.h>
//#include "esp_task_wdt.h"

extern "C" {
    #include "lwip/tcp.h"
};
#define MS_INTERVAL 500
#define BURST_SIZE 10
#define MSG_SIZE 30

struct tcp_pcb* pcb;
bool   cnx=false;

uint8_t msg[MSG_SIZE];

err_t _tcp_connected(void* arg, void* tpcb, err_t err){
  Serial.printf("Connected to echoserver\n");
  cnx=true;
  tcp_nagle_disable(pcb);
}

void setup(){
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin("XXXXXXXX", "XXXXXXXX");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }
  Serial.printf("\nIP: %s\n",WiFi.localIP().toString().c_str());
  pcb=tcp_new();
  IPAddress ip(192,168,1,20); // 192.168.1.20:8080 is an "echo server"
  ip_addr_t ipt;
  ip_addr_set_ip4_u32(&ipt, ip);
  err_t err = tcp_connect(pcb, &ipt, 8080,(tcp_connected_fn)&_tcp_connected);
}

uint32_t T;
int ticker=0;

void loop(){
  if(cnx && ((millis() - T) > MS_INTERVAL)){
    T=millis();
    if(++ticker > 59){
      Serial.printf("TICK\n",millis());
      ticker=0;
    }

    for(int i=0;i<BURST_SIZE;i++){
      if(MSG_SIZE < tcp_sndbuf(pcb)) tcp_write(pcb,(const uint8_t*) msg,MSG_SIZE,TCP_WRITE_FLAG_COPY);
//      else esp_task_wdt_reset(); // spin until space available
        else ESP.wdtFeed();    
    }
  }  
}
```

### Debug log (edited)

```
10:16:15.243 -> IP: 192.168.1.244
10:16:15.243 -> Connected to echoserver
10:16:44.822 -> TICK
10:17:14.861 -> TICK
... SNIP ...
12:47:03.301 -> TICK
12:47:33.374 -> TICK
12:48:03.410 -> TICK
12:48:33.463 -> TICK
12:49:03.534 -> TICK
```

philbowles, 2021-08-12:
I think the problem is that the rate of sending exhasts the internal queue, and that some kind of pause is required if tcp_write returns an error - which is of course not checked in this MCVE.

I will close this until I have done more research on the issue.

closed 2021-08-12 by philbowles