espressif/arduino-esp32 #5524: Setting ESP32 BLE Manufacturing ID 0x02E5

JonRobo, 2021-08-11:
I am trying to set the espressif manufacturer ID of 0x02e5 to my BLE GATT server. Having this ID present actually helps a lot in the connection of certain phones. I believe phones are now verifying the device they're connected to through this ID, if they don't like the ID they will disconnect.

I was digging through BLEAdvertising.cpp and found setManufacturerData(std::string data) and tried it out.
On my LightBlue app I could not see anything under 'Manufacturing specific data'.

I changed the source code to make the esp_ble_adv_data_t   m_advData have the espressif id by default

#define BLE_MANUFACTURER_DATA_LEN  	4
static uint8_t ROBO_BLE_MANUFACTURER_DATA[BLE_MANUFACTURER_DATA_LEN] = {0xE5, 0x02, 0x00, 0x00};

m_advData.manufacturer_len  = BLE_MANUFACTURER_DATA_LEN
m_advData.p_manufacturer_data =  ROBO_BLE_MANUFACTURER_DATA;

and added a function much like the others to access the private variable in m_advData.

void BLEAdvertising::setManufacturer(uint8_t *manufacturer_id, uint8_t length) {
	m_advData.manufacturer_len = length;
	m_advData.p_manufacturer_data = manufacturer_id;
} // setManufacturer

Perhaps I was using the setManufacturerData() function incorrectly, but it looks like this function does not directly change the
m_advData.manufacturer_len and m_advData.p_manufacturer_data parameters. Is it intended for manufacturer info of a final product using the ESP32?

I am trying to build an Arduino library for an existing product using the esp32 and would really appreciate an update to the source code to have espressif's BLE ID set by default and/or set via a function like the one posted above.

Looking forward to some thoughts on this issue

Thank-you

chegewara, 2021-08-13:
The manufacturer data is a special type of advertising data and can be anything you want, does not have to be manufacturer VID.
https://docs.silabs.com/bluetooth/2.13/code-examples/stack-features/adv-and-scanning/adv-manufacturer-specific-data

VojtechBartoska, 2022-03-30:
Hello @JonRobo, I'm closing as it was answered above. Kudos to @chegewara, If needed you can reopen the issue.

closed 2022-03-30 by VojtechBartoska