espressif/arduino-esp32 #5533: Rapid calls to tcp_output cause Guru Meditatation failure at random

philbowles, 2021-08-12:
### Hardware:

[image]

<B>Version</B> 1.0.6
<B>IDE name</B> Arduino IDE 1.8.15
<B>PSRAM enabled</B> no
<B>Computer OS</B> Windows 10

### Description:

Further to my previous issue #5520 (closed by me) I have narrowed down the bug. It seems to be a timing error when managing the LwIP send Q. I believe the problem in the previous sketch that it was failing to check the error return from tcp_write which returns the (very unhelpful) ERR_MEM when the internal Q is exceeded even though there is tons of free memory and the total data sent are well within the TCP_WND value. Why LwIP has no ERR_Q is beyond me, however...That code runs 100% as expected on ESP8266 (with changed libs and wdt of course). 

I therefore assumed that LwIP had internal code for handling "out of Q" conditions and doing a non-blocking wait until the Q had space. On ESP32 this is either a) not the case or b) broken. On discovering this, I rewrote the MVCE to the one shown here, to highlight the obvious problem with ESP32 + LwIP send Q.

It sometimes works, sometimes doesn't: there is no obvious pattern to when it fails. When it does fail it reboots and will often then work. When it works, I manually RST the device for the next test

### Sketch:  

```cpp
#include<WiFi.h>
#include "esp_task_wdt.h"

extern "C" {
    #include "lwip/tcp.h"
};

#define BURST_SIZE 10
#define MSG_SIZE 500

struct tcp_pcb* pcb;
bool cnx=false;

uint8_t msg[MSG_SIZE];

err_t _tcp_connected(void* arg, void* tpcb, err_t err){
  Serial.printf("Connected to echoserver\n");
  cnx=true;
  tcp_nagle_enable(pcb);
}

void sendData(const uint8_t* data,size_t len){
    auto Tstart=millis(); 
    size_t  bytesSent=0;
    size_t  bytesRemaining=len;
    while(bytesRemaining){
        size_t available=tcp_sndbuf(pcb);
        if(available && (tcp_sndqueuelen(pcb) < TCP_SND_QUEUELEN )){ // if there is any space at all (and space in Q!), use it
            auto chunk=std::min(bytesRemaining,available); // send as much as will fit
            auto err=tcp_write(pcb,data+bytesSent,chunk,TCP_WRITE_FLAG_COPY);
            // error handling deliberately omitted
            Serial.printf("(e=%d) AVAILABLE=%d SENT=%d REMAINING=%d QL=%d\n",err,available,bytesSent,bytesRemaining,tcp_sndqueuelen(pcb));
            err=tcp_output(pcb);
            if(err) Serial.printf("ERROR! tcp_output returns %d\n",err);
            else {
              bytesSent+=chunk;
              bytesRemaining-=chunk;
            }
        } 
        else { // spin while LwIP clears buffers
          esp_task_wdt_reset();  
          yield(); // allow LwIP to clear buffers  
        }
    }
    Serial.printf("%u bytes sent in %u mS\n",bytesSent,millis()-Tstart);
}

void setup(){
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
[... 310 lines cut ...]
        print("RX ",len(data))
        print(data)

        if(data):
            self.send(data)

class EchoServer(asyncore.dispatcher):

    def __init__(self, host, port):
        asyncore.dispatcher.__init__(self)
        self.create_socket(socket.AF_INET, socket.SOCK_STREAM)
        self.set_reuse_addr()
        self.bind((host, port))
        self.listen(5)

    def handle_accept(self):
        pair = self.accept()
        if pair is not None:
            sock, addr = pair
            print ('Incoming connection from %s' % repr(addr))
            handler = EchoHandler(sock)

server = EchoServer('0.0.0.0', 8080)
asyncore.loop()
```

philbowles, 2021-08-13:
Further research shows this is very similar to:
`https://github.com/ambrop72/badvpn/issues/85`
and
`https://lists.gnu.org/archive/html/lwip-users/2015-01/msg00000.html`

While these are not ESP32, they are both caused by the same problem, and it seems the ESP32 implementation may be breaking the same rules... The above issues are due to an incorrect thread / context switch leading to two separate threads concurrently accessing LwIP which - from what I understand - is not thread-safe and _must_ run _all_  code from one context.

In the second of those links, one reply explains:

> Summary:

> - Do *NOT* call anything in interrupt context, nothing, never, never, 
use your OS semaphore signaling to an Ethernet/serial/… RX thread
> - memp_* functions are thread-safe if SYS_LIGHTWEIGHT_PROT is 
set, and again, thread safe does not mean it is interrupt safe, especially
if your hardware does nested interrupts
> - Do *NOT* call any function from the RAW API outside lwIP thread

It seems like the ESP32 is stepping outside those boundaries, since (obviouslly) all of the sketch code is called on the main thread and I believe that "under the hood" LwIP on the ESP32 runs in its own task...so my money is on a  badly-timed switch between the two. Given that the same code runs perfectly on ESP8266 where threads / contexts are not an issue, this tends to support that theory. I readily admit I'm "in over my head" at that point, so I fully accept it may be something else.

I have also introduced a delay(N) into the burst to put a time-gap between each successive call to try to find the "break point". Early indications are that if the timing is such that the Q rarely (if ever) exceeds 1 then the crash disappears. Once the Q goes above 1, the potential for the "random" crash reappears., irrespective of actual timing values. So while it _is_ a "timing issue" it is only so because the timing affects the Q growth and the _real_ problem is faulty Q handling.

FWIW, my conclusion is that there is a missing / incorrect / unsynchronised access to the tcp send queue by the "ack" process, i.e. that items are being both added to the Q from the main loop code of the sketch and simultaneously removed - without proper locking / synchronisation - by the ack process.

igrr, 2021-08-13:
@philbowles It looks like you are calling LwIP "Raw" API functions in your sketch, outside of LwIP thread. Could that be the reason? In ESP-IDF, LwIP "Raw" APIs shouldn't be called outside the LwIP thread, so are effectively off limits to the applications. There is a note about that [in the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html#netconn-api), but I suppose it's not made clear enough.

It is recommended to use BSD socket APIs instead.

philbowles, 2021-08-13:
@igrr Thanks for the reply. When I have put code inside lwip raw callbacks the tiny 6388 stack breaks almost any time I try to do anything "sensible" is there any easy way to increase that?

Are you saying that tcp_* calls simply cannot ever be used from main loop code on the ESP32?, ie that my whole strategy is doomed to failure?

Is the result of all this that it's impossible to write source code using LwIP raw API calls that is portable between ESP32 and ESP8266 ?

philbowles, 2021-08-13:
I hear you. It's quite clear - although it's what I'd call "pretty well buried" :) Thanks again for you help.

[image]

If your answers to my two previous questions are "Yes", "Yes" then obviously I will close the issue.

igrr, 2021-08-13:
> Are you saying that tcp_* calls simply cannot ever be used from main loop code on the ESP32?, ie that my whole strategy is doomed to failure?
> Is the result of all this that it's impossible to write source code using LwIP raw API calls that is portable between ESP32 and ESP8266 ?

The answer is "Yes", "Yes", unfortunately.

ESP-IDF itself [does use raw API](https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns_networking.c) in one place currently, but the concurrency issue is mitigated by doing every raw API function call [inside lwIP task](https://github.com/espressif/esp-idf/blob/a20df743f1c51e6d65b021ed2ffd3081a2feec64/components/mdns/mdns_networking.c#L296-L304) via some LwIP internal APIs. I definitely would not recommend doing this in your code — it tooks us a long time to weed out all the issues from this IPC approach, and we are considering replacing that code with a BSD socket based implementation which may be fractionally slower but a lot less fragile. However if your goal is to run the same code on the ESP8266 (non-OS SDK, arduino-esp8266) and ESP32 (ESP-IDF, arduino-esp32) then this approach also won't be of much use as `tcpip_api_call` isn't present in the ESP8266 (non-OS SDK, arduino-esp8266) version of LwIP.

philbowles, 2021-08-13:
@igrr. Thank you. It explains everything (even though its very depressing! :) ) I have already decided to go down a 2x separate code base path since my ESP8266 implemenation is very stable. Ironically so is the ESP32 compile until the stress testing runs!.

I now know why it's called "stress" testing.

Thanks again for your thorough, clear and helpful explanation and advice.

closed 2021-08-13 by philbowles