espressif/arduino-esp32 #5534: Fat file system crashes if file system is corrupt

jgamble-simple, 2021-08-12:
If a file becomes corrupt on the FAT file system, then instead of throwing an errno, the call just crashes.

In our project we are just patching ff.c like so:
```
--- fatfs/src/ff.c
+++ fatfs/src/ff.c
@@ -3767,12 +3767,12 @@ FRESULT f_read (
 						clst = get_fat(&fp->obj, fp->clust);	/* Follow cluster chain on the FAT */
 					}
 				}
-				if (clst < 2) ABORT(fs, FR_INT_ERR);
+				if (clst < 2) ABORT(fs, FR_DISK_ERR);
 				if (clst == 0xFFFFFFFF) ABORT(fs, FR_DISK_ERR);
 				fp->clust = clst;				/* Update current cluster */
 			}
 			sect = clst2sect(fs, fp->clust);	/* Get current sector */
-			if (sect == 0) ABORT(fs, FR_INT_ERR);
+			if (sect == 0) ABORT(fs, FR_DISK_ERR);
 			sect += csect;
 			cc = btr / SS(fs);					/* When remaining bytes >= sector size, */
 			if (cc > 0) {						/* Read maximum contiguous sectors directly */
@@ -3873,7 +3873,7 @@ FRESULT f_write (
 					}
 				}
 				if (clst == 0) break;		/* Could not allocate a new cluster (disk full) */
-				if (clst == 1) ABORT(fs, FR_INT_ERR);
+				if (clst == 1) ABORT(fs, FR_DISK_ERR);
 				if (clst == 0xFFFFFFFF) ABORT(fs, FR_DISK_ERR);
 				fp->clust = clst;			/* Update current cluster */
 				if (fp->obj.sclust == 0) fp->obj.sclust = clst;	/* Set start cluster if the first write */
@@ -3887,7 +3887,7 @@ FRESULT f_write (
 			}
 #endif
 			sect = clst2sect(fs, fp->clust);	/* Get current sector */
-			if (sect == 0) ABORT(fs, FR_INT_ERR);
+			if (sect == 0) ABORT(fs, FR_DISK_ERR);
 			sect += csect;
 			cc = btw / SS(fs);				/* When remaining bytes >= sector size, */
 			if (cc > 0) {					/* Write maximum contiguous sectors directly */

```

And then it no longer asserts on a corrupt file, and the program can then determine what it needs to do after inspecting the errno after calling an io function.

VojtechBartoska, 2022-03-30:
Hello @rishadali , sorry for late reply. Are you able to test your issue on development version 2.0.3-RC1 to check if this is still valid?. You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

jgamble-simple, 2022-03-30:
Yes....this no longer asserts on a corrupt file system.....I am using arduino 2.0.0 and we no longer have our custom patch in place to fix it.

closed 2022-03-30 by jgamble-simple