espressif/arduino-esp32 #5536: arduino-esp32/esp-idf and v1.0/v2.0 version retrieval differences

drws, 2021-08-12:
_This is probably more of a feature request than a bug report, concerning both v1.0 and v2.0 branches._

[`Esp.h`](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Esp.h) provides a runtime function `getSdkVersion()`, a wrapper for `esp-idf`'s `esp_get_idf_version()`, which reads version data from [`esp_idf_version.h`](https://github.com/espressif/esp-idf/blob/master/components/esp_common/include/esp_idf_version.h) containing `ESP_IDF_VERSION_*` macros.

A similar story is `arduino-esp32` version, the [`esp_arduino_version.h`](https://github.com/espressif/arduino-esp32/commits/master/cores/esp32/esp_arduino_version.h) providing `ESP_ARDUINO_VERSION_*` macros. It has been added only [recently](https://github.com/espressif/arduino-esp32/commits/master/cores/esp32/esp_arduino_version.h) to v2.0 branch only. **Firstly**, I propose `esp_arduino_version.h` is also ported to v1.0 codebase so the two major Arduino SDK versions are compatible in such basic regard.

However, the `arduino-esp32` does not contain `getSdkVersion()`-analogue function containing the version string, available at runtime. A `getSdk2Version()` if you will. **Second** proposal of this issue is that such function is added to v2.0 and v1.0 branches.

This would also make `arduino-esp32` version as accessible as `esp-idf` one is, without needing to include `esp_arduino_version.h`. Such basic information should probably be most easily accessible.

VojtechBartoska, 2022-09-21:
**Status update of this issue:**
- we have now version in corresponding headers
- we need to check matches and what we can do next (need review)

VojtechBartoska, 2022-11-21:
We now have headers and files with all versions, but there is nothing we can do for older released versions.

closed 2022-11-21 by VojtechBartoska