espressif/arduino-esp32 #5537: IRAM_ATTR loopTask() and/or inline loop()?

drws, 2021-08-13:
### Description:

The [current](url) `loopTask()` declaration `void loopTask(void *pvParameters);` does not use the [`IRAM_ATTR`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/memory-types.html) attribute. `IRAM_ATTR` loads code into RAM and makes it stay there. Since `loopTask()` is mostly a wrapper for `loop()` and is constantly running, **firstly** I believe its functionality should be permanently ready in (I)RAM and execution pipeline should not bother with its optimization. It is probably never _swapped_ in the current implementation, but I think the suggestion would make a cleaner one.

The **second**, related question is whether the `loop()` function should be marked [`inline`](https://en.wikipedia.org/wiki/Inline_function)? `inline` saves the function call overhead, which should be a noticeable part of `loopTask()` overhead, even if tiny on an absolute scale. Core 1 should be 100% user's.

The **final** question is what would happen if we did both? Would `loop()` become implicitly `IRAM_ATTR`'d? This is probably not desired, so it should be thought after in advance. I checked and the thing at least compiles in both cases, whether the inlined function called from an `IRAM`'d one also has an `IRAM` attribute or not.

igrr, 2021-08-13:
Hi @drws,

Regarding the first question, unless you see a specific case where lack of IRAM_ATTR on `loop_wrapper` is causing issues, I wouldn't recommend forcing this function to be in IRAM. Moving code into IRAM reduces the available amount of DRAM that can be used as heap memory (on ESP32-S3, ESP32-C3 and later chips). Of course, loop_wrapper function is small and likely won't make a lot of difference, but in the same way it wouldn't matter much for execution speed if it is placed into Flash.

Regarding making `loop` inline as you show, this wouldn't work in C, unfortunately, as the inline definition of the function needs to be in the header file and this header file has to be included into the source file where the function is used. In other words, if `main.cpp` file in the arduino-esp32 project calls `loop`, then `inline void loop() { ... }` needs to be defined in some header file which is included into `main.cpp`. Since the sketch file is not included (by the preprocessor) into main.cpp, inlining won't happen. There is another technique, which could have actually helped "inline" the loop function into loop_wrapper, called Link Time Optimization (LTO). However LTO is not currently supported in ESP-IDF or arduino-esp32, due to issues related to (surprise!) forced IRAM sections placements.

Regarding the third question, indeed if func1 is inlined it will end up in IRAM, because the caller is placed into IRAM. This is usually not a big issue, because the only downside is static code size increase, and reduced amount of DRAM. However this issue also happens the other way around, and does occasionally lead to bugs:
```c++
static void IRAM_ATTR func1()
{
    // this function disables the cache, does something, then re-enables it.
    // Since it will be executed while cache is disabled, it needs to be in IRAM,
    // so the developer has added IRAM_ATTR.
    spi_flash_disable_cache();
    (code...)
    spi_flash_enable_cache();
}

void wrapper() {   // not in IRAM
  func1();   // call func1
}
```

In this case the compiler may notice that `func1` is static, so it can't be called from other translation units, and it will inline its code into `wrapper`, even though there is no `inline` keyword. In this case, the `func1` code gets placed into Flash, since `wrapper` doesn't have IRAM_ATTR. As a result, the function will crash after the cache gets disabled, since the code between `spi_flash_disable_cache` and `spi_flash_enable_cache` won't be accessible.

So as you see the section placements can be pretty tricky, and should in general be avoided, unless there is a reason why the problem can't be solved by other means.

drws, 2021-08-29:
Hi @igrr,

What I've failed to emphasize was that my suggestions were primarily focused on the original ESP32 with half-a-megabyte of RAM. I don't see a specific issue without the IRAM_ATTR on `loopTask()`. The logic is that `loopTask` runs so frequently (usually it is the only core 1 task) that not having it RAM-bound could only waste resources by swapping out this small piece of code, which does very likely not happen anyway. The same does not apply for less capable newer ESP32 chips.

Thank you for the explanation regarding inline definitions, LTO and an insightful code example. Since sketch files aren't included by the preprocessor, then also `IRAM_ATTR`-tagging the `loop()` in `.ino` sticks to `Arduino.h` declaration and doesn't produce section placement?

VojtechBartoska, 2022-03-30:
Hello @drws, we will take a look on this again. Sorry for late reply.

igrr, 2022-04-05:
> What I've failed to emphasize was that my suggestions were primarily focused on the original ESP32 with half-a-megabyte of RAM. I don't see a specific issue without the IRAM_ATTR on loopTask()

Funny you say this, in fact on the original ESP32 it is very easy to overflow IRAM when Bluetooth and Wi-Fi are both enabled. So saving every bit of IRAM actually does make sense. I think we won't implement this proposal because there is very little to gain from placing such a small function into IRAM explicitly. Most of the time, it will stay in cache.

closed 2022-04-05 by igrr

drws, 2022-04-06:
@igrr Thank you for including the proposal in relevant discussions. The resolution makes sense. There is still one specific question remaining from above for you to confirm, as follows:

> Since sketch files aren't included by the preprocessor, then also `IRAM_ATTR`-tagging the `loop()` in `.ino` sticks to `Arduino.h` declaration and doesn't produce section placement?

Half a meg of RAM is still not much in comparison with majority of devices ESP32 communicates with through Wi-Fi or Ethernet. However, it should still be enough for running both Wi-Fi, Bluetooth and the relevant services with just a bit more software and firmware work (tunable or at least tuned memory regions, lighter BT stack, and such).