espressif/arduino-esp32 #5539: Could you add an example of BLE mesh to the Arduino

anshilike, 2021-08-13:
I strongly recommend adding BLE mesh, which is found in ESP-IDF, but it is troublesome to install this software. As a novice, I use arduino a lot, but there are no ble mesh examples in The Arduino. Could you please add BLE mesh to the Arduino?

chegewara, 2021-08-13:
You cant use BLE mesh with arduino-ide:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig#L405

Maybe you can do it using nimBLE, but im not sure.

lybgo, 2021-09-04:
need an example of  BLE mesh too

arihantdaga, 2022-01-02:
Is it available now ? 
How can we use some code examples shown in esp idf in our arduino projects. 
I tried to include these files into my Arduino projects- 
```
#include "esp_ble_mesh_defs.h"
#include "esp_ble_mesh_common_api.h"

```
But it throws error saying cant find these libraries

arihantdaga, 2022-04-17:
It's relavant. We need to implement BLE Mesh example for Arduino.

VojtechBartoska, 2022-04-20:
Adding this to Documentation Roadmap.

mnour94, 2023-02-02:
Are there any updates on this? A BLE mesh implementation in Arduino would be greatly appreciated

VojtechBartoska, 2023-08-23:
Can you take a look on this @lucasssvaz?

Current list of BLE examples is here: https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE/examples

lucasssvaz, 2023-08-31:
There's no BLE mesh library for now. A makeshift solution is required for the time being.

I modified some files to be able to properly compile and run the [ESP-IDF example](https://github.com/espressif/esp-idf/tree/82cceabc6e/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server). Note that it requires to manually recompile the libraries to add support for specific BLE Mesh functions (this isn't trivial if you haven't done it before).

https://gist.github.com/lucasssvaz/bd105988aaf72ec291ecf94ecc2c692d

skyfly200, 2023-09-21:
Thanks for your help on this @lucasssvaz!
I too am looking to use this functionality in Arduino

VojtechBartoska, 2023-12-21:
Closing this as solved for now as @lucasssvaz provided workaround for IDF examples. For now, this is a first step, should be later on replaced by proper BLE Mesh library.

closed 2023-12-21 by VojtechBartoska

verdiluca, 2024-03-27:
hi, @lucasssvaz @skyfly200 @VojtechBartoska  I'm trying to compile the project of @lucasssvaz, but I saw that you have to modify the libraries but I didn't understand how to do it, would you be able to explain it to me? Thank you

lucasssvaz, 2024-03-28:
@verdiluca Check https://github.com/espressif/esp32-arduino-lib-builder. You can edit the Sdkconfig files to suit your needs.

dtsitses, 2024-04-06:
Hi @lucasssvaz @skyfly200 @VojtechBartoska. I'm trying to compile the [project](https://gist.github.com/lucasssvaz/bd105988aaf72ec291ecf94ecc2c692d) of @lucasssvaz but I'm getting multiple compilation errors. Checking the recommended lib builder [here](https://github.com/espressif/esp32-arduino-lib-builder) does not help as it's unclear what configuration changes are required, and also how the recompiled lib is referenced from the ino project which seems to still want to use the standard one installed in Arduino IDE via the board manager.

Would it be possible to make a step-by-step guide so that we can get this to work? Many thanks!

lucasssvaz, 2024-04-08:
@dtsitses Sorry about that, I've updated the links in the gist to use the 5.1 tag of IDF. You should be able to find the options needed now. Also, we just added an UI to the lib builder. Feel free to try that (although it is probably still harder than just using Arduino as a component and changing the IDF settings).