espressif/arduino-esp32 #5542: flash partition for esp32 dev board with 16MB and binary compiled under arduino ide

tradexsrl, 2021-08-14:
Hi i'm using arduino ide 1.8.13 and added the latest esp32 core , if i buy a 16MB esp32 board how can i choose to have for example 8 MB for app and 8 MB for SPIFFS?
i need to have OTA.
this problem came out because i'm using a ttgo esp32 dev board with 4MB of flash but the program i wrote for an arduino mkrgsm1400 board , adapted for esp32 , became over 1200kb (and it doesn't contain all the feature i had in mkrgsm1400...still under development) instead of 80kb when compiled....why ? 
library used are:
<TinyGsmClient.h>
"WiFi.h"
<esp_wifi.h>
<esp_bt.h>
"driver/adc.h"
<Adafruit_Sensor.h>
<DHT.h>
<DHT_U.h>
<ArduinoMqttClient.h>
"HX711.h"
"soc/rtc.h"
<ArduinoOTA.h>
<ArduinoHttpClient.h>

in the program for arduino mkr:
<MKRGSM.h>
<Adafruit_Sensor.h>
<DHT.h>
<DHT_U.h>
<ArduinoUniqueID.h>
<ArduinoMqttClient.h>
<ArduinoLowPower.h>
<Arduino_MKRGPS.h>
"HX711.h"
<ArduinoOTA.h>
<ArduinoHttpClient.h>

lbernstone, 2021-08-14:
esp32 has an operating system, along with large wifi & bluetooth drivers.  It will always make much larger binaries than running on bare metal.
Read through [this issue](https://github.com/espressif/arduino-esp32/issues/4551) for some pretty clear instructions how to partition.

VojtechBartoska, 2022-03-30:
Hello @tradexsrl, did you solve your issue?

VojtechBartoska, 2022-03-30:
Ok, so then I will close this issue and if needed, open a new one as there is a issue template which gives us more information. Thanks for your quick response and good luck with your projects.

closed 2022-03-30 by VojtechBartoska