espressif/arduino-esp32 #5543: bug: SPIFFS_Time example for Arduino not always reporting correct time, and sometimes SPIFFS errors

societyofrobots, 2021-08-14:
hardware: ESP32 - [HUZZAH32](https://learn.adafruit.com/adafruit-huzzah32-esp32-feather), with no additional hardware attached
Fresh unit, never before used, so nothing previously stored in SPIFFS.
1.0.6
Arduino IDE

My code: [SPIFFS_time](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/examples/SPIFFS_time/SPIFFS_time.ino) - unmodified, used as is without change

I ran the exact same code four different times, and got four different results.

**First time** I run the code, I only get this (correct time but SPIFFS fail):
```
Connecting to X_Mobile
............WiFi connected
IP address: 
192.168.10.137
Contacting Time Server

Now is : 2021-08-15 01:37:45

E (8794) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
Card Mount Failed
```

**Second time** I run the code, I get this (wrong time, SPIFFS fail):
```
Connecting to X_Mobile
............WiFi connected
IP address: 
192.168.10.137
Contacting Time Server

Now is : 1970-01-01 01:00:09

E (8794) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
Card Mount Failed
```

I forgot to save the output, but the **third time** I ran it, it said it saved files successfully. No issue, other than the date of 1970 was wrong.

**Fourth time** I run it, I get this (right time, different SPIFFS fail):

```
Connecting to X_Mobile
.....WiFi connected
IP address: 
192.168.10.144
Contacting Time Server

Now is : 2021-08-15 01:32:30

Listing directory: /
  FILE: /test.txt  SIZE: 150528  LAST WRITE: 1970-01-01 01:00:03
Removing Dir: /mydir
[E][vfs_api.cpp:188] rmdir(): rmdir is unnecessary in SPIFFS
Dir created
Deleting file: /hello.txt
[E][vfs_api.cpp:135] remove(): /hello.txt does not exists or is directory
Delete failed
Writing file: /hello.txt
[E][vfs_api.cpp:270] VFSFileImpl(): fopen(/spiffs/hello.txt) failed
Failed to open file for writing
Appending to file: /hello.txt
[E][vfs_api.cpp:270] VFSFileImpl(): fopen(/spiffs/hello.txt) failed
Failed to open file for appending
Listing directory: /
  FILE: /test.txt  SIZE: 150528  LAST WRITE: 1970-01-01 01:00:03
```

societyofrobots, 2021-08-15:
addendum...

I then tried [SPIFFS_Test](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/examples/SPIFFS_Test/SPIFFS_Test.ino), unmodified, and SPIFFS seems to be working fine with that example program. So, I don't think it's my hardware or setup, just SPIFFS_time that has the issue.

My result:
```
......- file written
Appending to file: /hello.txt
- message appended
Reading file: /hello.txt
- read from file:
World!
Renaming file /hello.txt to /foo.txt
- file renamed
Reading file: /foo.txt
- read from file:
World!
Deleting file: /foo.txt
- file deleted
Testing file I/O with /test.txt
- writing.............
```

VojtechBartoska, 2022-03-30:
Hello, sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska