espressif/arduino-esp32 #5544: There is a problem with the ESP32 timer, and there is no response when the program is downloaded

mjkdz, 2021-08-15:
There is a problem with the ESP32 timer, and there is no response when the program is downloaded

/*
 * ESP32共有4个定时器0-3。这些都是基于64位的通用定时器。
 * 每个定时器都有一个16位预分频器（2-65536)以及可以自动
 * 重新加载的 64 位递增/递减计数器
*/
hw_timer_t*timer = NULL;//配置定时器我们需要一个指向hw_timer_t类型变量的指针
volatile int led=2;
volatile byte ZT =0;//volatile关键字说明我们需要在中断中修改这个变量
void IRAM_ATTR kaiDS(){//定时器中断函数
  ZT=!ZT;
  digitalWrite(led,ZT);
}

void setup() {
  Serial.begin(115200);
  pinMode(led, OUTPUT);
  
  //设置定时器,0是定时器(4个定时器)80预分频器的值,true上升沿计数.false下降沿计数
  timer=timerBegin(0,80,true);
  //Timer指针变量,kaiDS触发中断执行函数,Timer 触发器与时钟同步true上升沿时触发定时器false当时钟信号改变电平时触发定时器
  timerAttachInterrupt(timer,&kaiDS,true);
  //Timer指针变量,1000000触发中断频率,true每次触发后自动重置定时器中断
  timerAlarmWrite(timer,1000000,true);
  timerAlarmEnable(timer);//启动定时器
}
void loop() {
}

mjkdz, 2021-08-15:
The previous version is good

VojtechBartoska, 2021-08-18:
@mjkdz Please can you rewrite your issue in English?

VojtechBartoska, 2021-11-18:
@mjkdz Do you have any update to this issue? If not, I will close it in few days.

closed 2021-12-23 by VojtechBartoska