espressif/arduino-esp32 #5550: LittleFS not working with ESP32-S2

Jason2866, 2021-08-16:
What you are trying to do?
Use LittleFS with the ESP32-S2 (the same example does work with the ESP32-C3 and ESP32)

Describe your system(ESP32-S2, Win10 latest, branch master).
Describe what is failing.
Error messages during initalizing LittleFS

Show the shortest possible code that will duplicate the error.
Example in lib folder `LittleFS` see bottom
Show the EXACT error message(it doesn't work is not enough).

### Hardware:
Board:							ESP32-S2 Dev Module
Core Installation version:			git master
IDE name:						Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                   no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
LittleFS does not initialize correctly and is not working
Sketch https://github.com/espressif/arduino-esp32/tree/master/libraries/LittleFS/examples/LITTLEFS_PlatformIO

Jason2866, 2021-08-24:
@me-no-dev do you need more infos or logs?

me-no-dev, 2021-08-25:
is this issue particular to platformio @Jason2866 ?

Jason2866, 2021-08-25:
No, i tried the example https://github.com/espressif/arduino-esp32/blob/master/libraries/LittleFS/examples/LITTLEFS_test/LITTLEFS_test.ino with Arduino IDE. Same error.

closed 2021-11-11 by Jason2866

Dario-Ciceri, 2023-01-11:
> @me-no-dev No, i tried the example https://github.com/espressif/arduino-esp32/blob/master/libraries/LittleFS/examples/LITTLEFS_test/LITTLEFS_test.ino with Arduino IDE. Same error.

did you fix it? i have this problem too with SPIFFS: does not start with /

lbernstone, 2023-01-11:
@Dario-Ciceri Please do not hijack issues.  Where do you see this error in the post?  If you have a problem with SPIFFS, open a new issue.

Dario-Ciceri, 2023-01-11:
> 

I'm sorry, I didn't know about hijacking... btw I fixed it was a problem due to a missing "/", still getting problems since I get error 6 as it can't read the file. I'm trying to display gifs on led matrix

oleksandr007, 2023-01-19:
I dont't really understand what was the solution to this issue? I have ESP32-S2 mini and im trying days to figure out how to make LittleFS work. The compiler say it's everything ok but ESP32 can't mount LittleFS.

Jason2866, 2023-03-15:
The issue is solved since the bootloader uses the correct flash mode. Safe to use since core 2.0.6

diegostf, 2023-03-15:
> The issue is solved since the bootloader uses the correct flash mode. Safe to use since core 2.0.6

Where can I find the settings/flash mode?