espressif/arduino-esp32 #5552: ESP32 C3 digitalRead() returns always 0, should return the pin level

helmut64, 2021-08-16:
When I do a digitalWrite() to set the GPIO pin (e.g. 2) to HIGH, a digitalRead(2) returns always 0
It should return the pin level as it does not the regular ESP32 or the S2
Arduino-esp32 version: 2.0.0-alpha1
Regards Helmut

atanisoft, 2021-08-16:
This is expected behavior as the pin is configured as output only. Reading an output pin will always return zero. The pin must be configured as INPUT_OUTPUT for the read to work. Unfortunately there is no Arduino API to configure a pin as INPUT_OUTPUT, you can however call:
```
gpio_set_direction(PIN, GPIO_MODE_INPUT_OUTPUT);
```
where PIN is the target GPIO number.

helmut64, 2021-08-17:
Hi Mike, thank you for that advise, your workaround works.
However I am using the Arduino API, digitalRead on the output set GPIO's works on Arduino (ESP32, Atmel-D21, ESP32-S2, etc). On Mbed OS this is also supported. I believe it would be great if the C3 behaves like the other boards/platforms.

Again, thank you for your advise.
PS: I am using like many others reading the GPIO and setting the inverted value.
Regards from Hannover Helmut

chegewara, 2021-08-17:
There is simple explanation. ESP32 C3 just does not support it.
You can use this simple sketch, which is using esp-idf API, to test it:
```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  pinMode(7, OUTPUT);
  digitalWrite(7, 0);
  delay(1000);
  Serial.print("pin level: ");
  Serial.println(gpio_get_level((gpio_num_t)7));
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  digitalWrite(7, 1);
  Serial.print("pin level: ");
  Serial.println(gpio_get_level((gpio_num_t)7));
  digitalWrite(7, 0);
  delay(1000);
  Serial.print("pin level: ");
  Serial.println(gpio_get_level((gpio_num_t)7));
}
```
Like you said, it works on S2 and does not work on C3, and it cant be changed or fixed in arduino as long as it wont be "fixed" in esp-idf. This also may be C3 hardware design and cant be fixed.

chegewara, 2021-08-17:
There is simple explanation. ESP32 C3 just does not support it.
You can use this simple sketch, which is using esp-idf API, to test it:
```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  pinMode(7, OUTPUT);
  digitalWrite(7, 0);
  delay(1000);
  Serial.print("pin level: ");
  Serial.println(gpio_get_level((gpio_num_t)7));
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  digitalWrite(7, 1);
  Serial.print("pin level: ");
  Serial.println(gpio_get_level((gpio_num_t)7));
  digitalWrite(7, 0);
  delay(1000);
  Serial.print("pin level: ");
  Serial.println(gpio_get_level((gpio_num_t)7));
}
```
Like you said, it works on S2 and does not work on C3, and it cant be changed or fixed in arduino as long as it wont be "fixed" in esp-idf. This also may be C3 hardware design and cant be fixed.

@igrr

helmut64, 2021-08-17:
Again, I understand that it is not working. Ff it can be fixed in the IDF, I would recommend to do so.

chegewara, 2021-08-17:
Then you should report it in esp-idf, not here, if you understand it.

helmut64, 2021-08-17:
But here is the place for the ESP32-C3-Arduino, and it is an Arduino behaviour. I will have a look into the IDF source to check out what is behind.

chegewara, 2021-08-17:
I did show you example code with plain esp-idf API, it is not arduino, which shows the same output:
`gpio_get_level((gpio_num_t)7)`

atanisoft, 2021-08-17:
This *CAN* work in arduino-esp32 but the support code for it does not exist yet. A new pin mode will be necessary (INPUT_OUTPUT or similar name) which is a combination of the current INPUT and OUTPUT modes. The underlying ESP-IDF code does not support reading the state of an OUTPUT only pin and that is why the current arduino code does not work as you might expect.

As for why this works for ESP32 and ESP32-S2 but not ESP32-C3 that is due to the ESP32-C3 code using ESP-IDF APIs whereas the others are using a local implementation which does not work for the ESP32-C3 (and long term should be removed entirely in favor of ESP-IDF APIs).

chegewara, 2021-08-17:
@atanisoft this is esp-idf API, which works on esp32 with pin OUTPUT only:
`gpio_get_level((gpio_num_t)7)`
the same API does not work on C3, so the problem is not arduino perse, and difference in esp-idf on different models.

atanisoft, 2021-08-17:
@chegewara it shouldn't work even in ESP-IDF unless the pin mode is INPUT_OUTPUT since the HAL code references the hardware IN register which is not updated as part of OUTPUT only pin manipulation.

chegewara, 2021-08-17:
Maybe it should not work, but it works. Either way it may be consider bug and should be reported in esp-idf, not here.

To be honest until now ive been thinking the same way, that reading OUTPUT pin should not be possible and i never before even did try it, since there is INPUT_OUTPUT option.

helmut64, 2021-08-17:
A common technic ist digitalWrite(!digitalWrite()) to invert the current output.
For me this was just once reason why existing software did not work with the C3.

lbernstone, 2021-08-17:
I think the expected <b>arduino</b> behavior here is that you can digitalRead an output pin.  That probably means the implementation is INPUT_OUPUT once the core is fully refactored on the idf hal.

atanisoft, 2021-08-17:
The ESP32-C3 uses ESP-IDF APIs but ESP32 and ESP32-S2 use direct register access. This issue should be used to standardize on ESP-IDF APIs *AND* default OUTPUT mode to be INPUT_OUTPUT.

chegewara, 2021-08-17:
@atanisoft are you sure? As far as i understand C3 is not using esp-idf, but i may be wrong:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-gpio.c#L324

atanisoft, 2021-08-17:
@chegewara Certain... https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-gpio.c#L41 This if/elif block enters this line when using the C3.

chegewara, 2021-08-17:
ok, i did some more testing
@helmut64 my bad, it looks like the problem is with arduino-esp32, not esp-idf, sorry

there is something odd with `pinMode`, because setting mode with esp-idf `gpio_set_direction` call and read value from OUTPUT only pin is consistent on S2, C3 and probably esp32 too, always 0.

helmut64, 2021-08-18:
I did some testing once again, it is only a C3 problem, the S2 or the regular ESP32 and other MCUs don't show this behaviour.
```
void setup() {
  Serial.begin(115200);
  pinMode(2, OUTPUT);
}

void loop() {
  digitalWrite(2, HIGH);
  Serial.printf("Pin: %d\n", digitalRead(2));
  delay(1000);
  digitalWrite(2, LOW);
  Serial.printf("Pin: %d\n", digitalRead(2));
  delay(1000);
}
```

VojtechBartoska, 2022-03-30:
Hello @helmut64 , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

helmut64, 2022-03-30:
Hello @VojtechBartoska the problem still exists with the 2.0.3-RC1 release

atanisoft, 2022-03-30:
@P-R-O-C-H-Y Change https://github.com/espressif/arduino-esp32/blob/1e388a24cef1b336990b8f67dbce63cc8a4794a4/cores/esp32/esp32-hal-gpio.h#L45 to `0x3` which maps to https://github.com/espressif/esp-idf/blob/e3c0229d2ec9bfd0203a3cab18d6642798457578/components/hal/include/hal/gpio_types.h#L376. Perhaps it would make more sense to use the IDF type values directly?

helmut64, 2022-03-30:
Dear @atanisoft I understand, however in Arduino ESP32, Atmel D21 and all others Arduino's the GPIO  OUTPUT reading is working. I watched that this compatibility problem exists only with the C3

atanisoft, 2022-03-30:
> this compatibility problem exists only with the C3

That is due to C3 being the only ESP32 variant that uses the IDF APIs so far. It is not a bug in so far as how the pin is configured, it is strictly OUTPUT only and IDF does not track the state for OUTPUT only pins. It must be configured as INPUT_OUTPUT for IDF APIs to track and return the state.

P-R-O-C-H-Y, 2022-04-12:
@atanisoft Do you think its best to change OUTPUT to 0x03 (to be input_output) or add INPUT_OUTPUT option separately?

helmut64, 2022-04-12:
As Arduino pinMode has no INPUT_OUTPUT, I believe when pinMode OUTPUT is specified it should handle it as GPIO_MODE_INPUT_OUTPUT. This would be compatible with other existing Arduino MCUs. @P-R-O-C-H-Y

P-R-O-C-H-Y, 2022-04-12:
@helmut64  I will make the change, but will do some tests before placing PR to make sure, changing that doesn't break anything :)

closed 2022-04-21 by me-no-dev

AcidCaos, 2022-05-20:
Hi, I recently updated from 2.0.2 to 2.0.3, and a `pinMode(16, OUTPUT)` on my code, which worked fine before, does not now: it makes my board reboot. I'm using a TTGO LoRa T3_V1.6.1. I've tried to put 0x02 instead of OUTPUT, but it still does not work, so not really sure if it is related with this commit.

chai2010, 2025-06-03:
Now that I'm using Arduino, do I still care about the little performance saved by GPIO? What I value is Arduino's compatibility and user experience. I've wasted a lot of time on the design of Arduino Nano ESP32, and now my impression of ESP32 is just like many Android phones – although their hardware specifications are impressive, the user experience is poor.

me-no-dev, 2025-06-03:
> Changing the value of OUTPUT to something else is absolutely a garbage design!

No change in design. @AcidCaos is probably using a board with PSRAM and that is connected to IO16 and IO17.

> I am writing this because Arduino boards (including ESP32 Xtensa MCUs) allow to read GPIO ports even if they are set to OUTPUT.

Issue introduced in 3.0 and fixed in the latest code