espressif/arduino-esp32 #5553: Error to connect to EVO192 with serial over ip

danilodivaio, 2021-08-16:
Hello to all!
I try to connect to alarm panel (paradox EVO 192 by serial over ip with a esp32) but I've a problem:

2021-08-14 11:11:11,871 - INFO - PAI.paradox.connections.ip.connection - Connecting. Try 1/3
2021-08-14 11:11:11,988 - INFO - PAI.paradox.connections.connection - Connection established
2021-08-14 11:11:11,990 - INFO - PAI.paradox.paradox - Connecting to Panel
2021-08-14 11:11:17,004 - ERROR - PAI.paradox.paradox - Timeout while connecting to panel. Is an other connection active?
2021-08-14 11:11:17,006 - ERROR - PAI - Unable to connect to alarm

my configuration:

LOGGING_LEVEL_CONSOLE: 20
LOGGING_LEVEL_FILE: 40
CONNECTION_TYPE: IP
SERIAL_PORT: /dev/ttyUSB0
SERIAL_BAUD: 57600
IP_CONNECTION_HOST: 192...*
IP_CONNECTION_PORT: 23
IP_CONNECTION_PASSWORD: paradox
KEEP_ALIVE_INTERVAL: 10
LIMITS:
zone: auto
user: 1-10
door: ''
pgm: 1-5
partition: auto
module: ''
repeater: ''
keypad: ''
key-switch: ''
SYNC_TIME: true
SYNC_TIME_MIN_DRIFT: 120
PASSWORD: '0000'
MQTT_ENABLE: true
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_KEEPALIVE: 60
MQTT_USERNAME: *********
MQTT_PASSWORD: **********
MQTT_HOMEASSISTANT_AUTODISCOVERY_ENABLE: true
COMMAND_ALIAS:
arm: partition all arm
disarm: partition all disarm
MQTT_COMMAND_ALIAS:
armed_home: arm_stay
armed_night: arm_sleep
armed_away: arm
disarmed: disarm
HOMEASSISTANT_NOTIFICATIONS_EVENT_FILTERS:

live,alarm,-restore
live,trouble,-clock
live,tamper
PUSHBULLET_CONTACTS: []
PUSHBULLET_EVENT_FILTERS:
live,alarm,-restore
live,trouble,-clock
live,tamper
PUSHOVER_EVENT_FILTERS:
live,alarm,-restore
live,trouble,-clock
live,tamper
PUSHOVER_BROADCAST_KEYS: []
SIGNAL_CONTACTS: []
SIGNAL_EVENT_FILTERS:
live,alarm,-restore
live,trouble,-clock
live,tamper
GSM_CONTACTS: []
GSM_EVENT_FILTERS:
live,alarm,-restore
live,trouble,-clock
live,tamper
IP_INTERFACE_ENABLE: false
IP_INTERFACE_PASSWORD: paradox
DUMMY_EVENT_FILTERS: []
IP_CONNECTION_BARE: true

how can i solve? Thanks to all!

VojtechBartoska, 2022-03-30:
Hello @danilodivaio, still actual?

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska