espressif/arduino-esp32 #5555: Multiple functions in classes that should be const are not const

PORTB, 2021-08-17:
This is relevant to the core code, not a specific board/hardware.

### Description:
There are multiple functions in classes that are not marked as const, primarily getters and toString().
For example, in `WiFi.h`
```cpp
bool isProvEnabled();
```
should be
```cpp
bool isProvEnabled() const;
```
and in `WiFi.cpp`
```cpp
bool WiFiClass::isProvEnabled()
{
    return prov_enable;
}
```
should be 
```cpp
bool WiFiClass::isProvEnabled() const
{
    return prov_enable;
}
```
This is also in other classes, such as `isConnected()`, `getMTU()`, `toString()` in BLEClient

VojtechBartoska, 2022-03-30:
Hello @PORTB, thanks for your contribution. We will consider this and let you know.

mrengineer7777, 2023-03-29:
Hi @PORTB .  How would this improve the library?

closed 2025-01-16 by Parsaabasi