espressif/arduino-esp32 #5557: SPIFFS upload fail

savagerex, 2021-08-18:
when i use "ESP32 Sketch Data Upload".

it happen error

as below

esptool.py v2.6
Serial port COM22
Connecting.....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: c4:dd:57:79:b0:e4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB

A fatal error occurred: File C:\Users\REXYH~1.LIU\AppData\Local\Temp\arduino_build_452606/SPIFFS_test.spiffs.bin (length 1638400) at offset 6750208 will not fit in 4194304 bytes of flash. Use --flash-size argument, or change flashing address.
SPIFFS Upload failed!

[image]

### Hardware:
Hardware:
Hardware:
Board : ESP32 dev module
Core install version: 1.0.4
IDE name : Arduino
Flash Frequency : 80MHz
PSRAM Enable : yes
upload speed :115200
computer os : win10

#include "SPIFFS.h"

void setup()
{
  Serial.begin(115200);

  if(!SPIFFS.begin(true))
  {
     Serial.println("error");
     return;  
  }

  SPIFFS.format();

   File file = SPIFFS.open("/test_ca.txt");

   if(!file)
   {
      Serial.println("fail");
      return; 
   }

   Serial.println("Content");

   while(file.available())
   {
       Serial.write(file.read());   
   }

   file.close();
}

void loop()
{
  
}

AdamMiltonBarker, 2021-08-22:
I have had this issue, hold the boot button while your sketch is compiling and it will connect, if that fails then revert the board to 1.0.4 and repeat, this was happening to me today and that is how I solved it.

AdamMiltonBarker, 2021-08-22:
Your file is too big.

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska