espressif/arduino-esp32 #5558: ESPLwIPClient setTimeout conflicts with Stream setTimeout

jandrassy, 2021-08-18:
ESPLwIPClient::setTimeout conflicts with Stream::setTimeout

the classes hierarchy is Print <- Steam <- Client <- ESPLwIPClient <- WiFiClient 

https://github.com/espressif/arduino-esp32/blob/b580bb23fdcce0d3063b0b0f766a74e32bd8ea67/cores/esp32/Stream.h#L61

https://github.com/espressif/arduino-esp32/blob/b580bb23fdcce0d3063b0b0f766a74e32bd8ea67/libraries/WiFi/src/WiFiClient.h#L36

VojtechBartoska, 2022-03-30:
Hello @jandrassy, sorry for late reply. We will take a look on this if we can fix this.

P-R-O-C-H-Y, 2022-04-25:
Hi @JAndrassy , @rin67630 
Can you provide some sketch, so I can test and fix that? :) Thanks

JAndrassy, 2022-04-25:
so you should not change the Arduino API (Stream.h). I don't know what for is the other setTimeout but it should be removed or renamed

P-R-O-C-H-Y, 2022-04-25:
Sure the change will be in ESPLwIPClient class or WiFiClient class.
But @rin67630 mentioned that the timeout does not have any effect. So I would like to test that to be able to fix everything at once.

JAndrassy, 2022-04-25:
> Sure the change will be in ESPLwIPClient class or WiFiClient class. But @rin67630 mentioned that the timeout does not have any effect. So I would like to test that to be able to fix everything at once.

I guess rin's problem is that it has effect on something else than expected

P-R-O-C-H-Y, 2022-05-02:
@JAndrassy How did you compile that? I have no errors / warning during compilation and the timeout is working fine for me.

P-R-O-C-H-Y, 2022-05-02:
Can you test WifiClient.cpp and WifiClient.h files from my fork? I'm making changes now, because the virtual cannot be there :)

[link to by repo + branch with fixes](https://github.com/P-R-O-C-H-Y/arduino-esp32/tree/WifiClient_settimeout_conflict)

P-R-O-C-H-Y, 2022-05-02:
> Can you test WifiClient.cpp and WifiClient.h files from my fork? I'm making changes now, because the virtual cannot be there :)
> 
> [link to my repo + branch with fixes](https://github.com/P-R-O-C-H-Y/arduino-esp32/tree/WifiClient_settimeout_conflict)

You can test changes from this [PR](https://github.com/espressif/arduino-esp32/pull/6676)

JAndrassy, 2022-05-02:
sorry, I assumed you know the Arduino API Print and Stream classes as I think it is a very important part of Arduino core.

setTimeout and _timeout exist in Stream.h/cpp. _timeout is used in timed read functions readBytes, readBytesUntil, readString, readStringUntil, parseInt, parseFloat, find.
the _timeout in Stream class can't be set if setTimeout and _timeout exist in WiFiClient

P-R-O-C-H-Y, 2022-05-02:
Can you check latest changes in linked PR? setTimeout is removed from WifiClient / WifiClientSecure.

P-R-O-C-H-Y, 2022-05-02:
the setTimeout is for read/write functions. This change will make the timeouts to be set in milliseconds to be same as Arduino Stream setTimeout. The connection timeout can be set in call Client.connect() function eg. 
`int connect(IPAddress ip, uint16_t port, int32_t timeout);`

podaen, 2022-05-03:
If I understand it good the client timout needs to be the same as the timout of the stream?

Hi, I am testing it to but I can't get info of the stream.h with Serial.println();

podaen, 2022-05-03:
Having troubles to analyse the the stream input. I want to serial.print the count of the readed bytes in the stream.cpp but nothing is shown. If I serial.print in stream.h I got an error...

Stream.h: 104:9: error: 'Serial' was not declared in this scope
   Serial.print("read bytes stream.h")

P-R-O-C-H-Y, 2022-05-03:
> Having troubles to analyse the the stream input. I want to serial.print the count of the readed bytes in the stream.cpp but nothing is shown. If I serial.print in stream.h I got an error...
> 
> Stream.h: 104:9: error: 'Serial' was not declared in this scope
>    Serial.print("read bytes stream.h")

Instead of `Serial.print(read bytes stream.h)` use `log_d(read bytes stream.h)` and set Debug output level to DEBUG.

podaen, 2022-05-03:
Can't get that ether...

> HardwareSerial.h:50: In file included from
> HardwareSerial.cpp:7: from
> Stream.h: In member function virtual size_t Stream::readBytes(uint8_t*, size_t)
>  
> xtensa-esp32-elf-g++*: error: E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32\Debug\build_opt.h: No such file or directory
>  
> Stream.h: 104:9: error: 'log_d' was not declared in this scope
>    log_d("read bytes stream.h")
>    ^~~~~
> Stream.h:104: note  the macro log_d had not yet been defined

podaen, 2022-05-03:
Here the log after including #include "esp32-hal-log.h"

```
Compiling debug version of 'player-url-i2s' for 'DOIT ESP32 DEVKIT V1                                                                                                       (espressif_esp32doit-devkit-v1)'
C:\Program Files (x86)\Arduino\java\bin>chcp 65001  1>NUL 2>NUL 
C:\Program Files (x86)\Arduino\java\bin>if exist "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\partitions.csv" del /f "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\partitions.csv" 
C:\Program Files (x86)\Arduino\java\bin>chcp 65001  1>NUL 2>NUL 
C:\Program Files (x86)\Arduino\java\bin>if exist "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\player-url-i2s\player-url-i2s\partitions.csv" COPY /y "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\player-url-i2s\player-url-i2s\partitions.csv" "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\partitions.csv" 
C:\Program Files (x86)\Arduino\java\bin>chcp 65001  1>NUL 2>NUL 
        1 bestand(en) gekopieerd.
C:\Program Files (x86)\Arduino\java\bin>if not exist "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\partitions.csv" if exist "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\hardware\espressif\esp32\variants\doitESP32devkitV1\partitions.csv" COPY "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\hardware\espressif\esp32\variants\doitESP32devkitV1\partitions.csv" "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\partitions.csv" 
C:\Program Files (x86)\Arduino\java\bin>chcp 65001  1>NUL 2>NUL 
C:\Program Files (x86)\Arduino\java\bin>if not exist "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\partitions.csv" COPY "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\hardware\espressif\esp32\tools\partitions\partitions.csv" "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\partitions.csv" 
C:\Program Files (x86)\Arduino\java\bin>chcp 65001  1>NUL 2>NUL 
C:\Program Files (x86)\Arduino\java\bin>IF EXIST "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\player-url-i2s\player-url-i2s\bootloader.bin" (COPY /y "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\player-url-i2s\player-url-i2s\bootloader.bin" "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\player-url-i2s.ino.bootloader.bin"  )  ELSE (IF EXIST "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\hardware\espressif\esp32\variants\doitESP32devkitV1\bootloader.bin" (COPY "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\hardware\espressif\esp32\variants\doitESP32devkitV1\bootloader.bin" "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\player-url-i2s.ino.bootloader.bin"  )  ELSE (COPY /y "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\hardware\espressif\esp32\tools\sdk\esp32\bin\bootloader_dio_80m.bin" "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\player-url-i2s.ino.bootloader.bin"  ) ) 
        1 bestand(en) gekopieerd.
C:\Program Files (x86)\Arduino\java\bin>chcp 65001  1>NUL 2>NUL 
C:\Program Files (x86)\Arduino\java\bin>if exist "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\player-url-i2s\player-url-i2s\build_opt.h" COPY /y "G:\Software_Projects\Visual Studio\Visual Studio 2019\Arduino\player-url-i2s\player-url-i2s\build_opt.h" "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\build_opt.h" 
C:\Program Files (x86)\Arduino\java\bin>chcp 65001  1>NUL 2>NUL 
C:\Program Files (x86)\Arduino\java\bin>if not exist "E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\build_opt.h" type nul  1>"E:\AppData\Local\Temp\VMBuilds\player-url-i2s\espressif_esp32doit-devkit-v1\Debug\build_opt.h" 
 
esp32-hal.h:76: In file included from
Arduino.h:36: from
Stream.cpp:23: from
Stream.cpp: In member function virtual size_t Stream::readBytes(char*, size_t)
 
Stream.cpp: 293:11: error: expected ')' before 'count
   log_d(count)
   ^~~~~
esp32-hal-log.h:91: note  in definition of macro ARDUHAL_LOG_FORMAT
   %u] %s(): " format ARDUHAL_LOG_RESET_COLOR "\r\n", (unsigned long) (esp_timer_get_time() \ 1000ULL), pathToFileName(__FILE__), __LINE__, __FUNCTION__
   ^~~~~~
Stream.cpp:293: note  in expansion of macro log_d
   log_d(count)
   ^~~~~
 
esp32-hal.h:76: In file included from
Error compiling core
Arduino.h:36: from
Stream.cpp:23: from
Debug build failed for project 'player-url-i2s'
esp32-hal-log.h:111: note  to match this (
   #define log_d(format, ...) log_printf(ARDUHAL_LOG_FORMAT(D, format), ##__VA_ARGS__)
Stream.cpp:293: note  in expansion of macro log_d
   log_d(count)
   ^~~~~
```

JAndrassy, 2022-05-03:
@podaen why do you hijack my issue report for an unrelated problem?

podaen, 2022-05-03:
Why making a new one?? If I had that same issue, I want to see if it works for me too.

podaen, 2022-05-03:
@P-R-O-C-H-Y  Can you do this for [client.h ](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Client.h)as well?

VojtechBartoska, 2022-05-03:
> Why making a new one?? If I had that same issue, I want to see if it works for me too.

Hi @podaen, up to your comment it seems you are facing different issue (correct me if I'm wrong). Due to that it's better to open an new issue if it's needed. The goal is to uncover specific problem in particular issue and solved them directly.

Other benefit of opening a new issue is providing more complex information thanks to the issue template. From a single comment it's almost impossible to just reproduce your code and at least verify the issue.

Also, if you are more experienced and as I can see you have specific request, don't hesitate to open a Pull Request. Generally it's much faster approach comparing to opened issue.

I hope you understand all parts mentioned above. We would like to help you as much as possible but please keep in mind that are resources are also limited.

Thanks.

podaen, 2022-05-03:
Why so defendant? I admit that it was wrong to print in the stream over the hardware serial. I investigated it and it was is not the right thing to do... My attention didn't came out of nowhere, it came out of the stream and timeout. And I make some correction and I will test too!

podaen, 2022-05-03:
@JAndrassy The socket will have the timeout when connecting to the wificlient like you wanted. So, you can close this topic. 

Still under investigation: The timeout is not been shared from the wificlient or the sll client to the stream.h. The client.h is the link for sharing to stream.h! I have to keep this in mind to continue tomorrow.

JAndrassy, 2022-05-03:
@podaen I created this issue report not for some troubleshooting, but to resolve a big mistake in the esp32 WiFi library which can't be resolved without a breaking change and a decision must be made about it by maintainers. 
your comments are out of context here. there is nothing to debug (and learn how to debug)

podaen, 2022-05-03:
Read my post.

VojtechBartoska, 2022-05-04:
Issue postponed to 2.1.0 milestone. More investigation and testing is needed. Thanks for understanding.

JAndrassy, 2023-10-15:
Other libraries have [client.setConnectionTimeout](https://www.arduino.cc/reference/en/libraries/ethernet/client.setconnectiontimeout/). It was first used in the 2.0.0 version of the Ethernet library in 2018. There it is used in client.stop() too.

PR #6676 (2023-12-13): ESPLwIPClient::setTimeout conflict fix with Stream::setTimeout
## Description of Change
Arduino Stream::settimeout() is not virtual so it cannot be virtual in ESPLwIPClient.
This change should fix this conflict.

## Tests scenarios

## Related links
Closes #5558
closed 2023-12-13 by me-no-dev