espressif/arduino-esp32 #5559: ADC has range problems in new dev branchs

saeedbala, 2021-08-18:
Hi! I just discovered in the new versions, the ADC ref/attenuation is messed up.
Connecting a 1.65 volt to ADC saturates the values to 4095. 
If I decrease the voltage to below 1 Volt, it shows the values that if the reference was connected to a 1 Volt source. So for example a 0.5 Volt voltage has a reading of 2000/4095. 
I tried to set the attenuation manually too but still doesn't work.
This code works perfectly on the stable libraries while it shows saturated values on Dev ones (2..0.0RC): 

### Hardware:
Board:							node32
Core Installation version:			2.0.0
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Describe your problem here
the ADC ref/attenuation is messed up.

### Sketch:  
```cpp

//Change the code below by your sketch
void setup() 
{
 
  Serial.begin(115200);
  
  Serial.println("HELLO!");
  analogReadResolution(12);
  analogSetAttenuation(ADC_11db);
 
}
 
void loop() 
{
  Serial.println(analogRead(36));
  delay(500);
}
```

### Debug Messages:
```
HELLO
4095
4095
4095
```

felmue, 2021-08-18:
Hello @saeedbala 

have a look [here.](https://github.com/espressif/arduino-esp32/issues/5503)

Thanks
Felix

marcboon, 2021-08-24:
For the moment use: analogSetPinAttenuation(pin, ADC_11db) for each pin you intend to use with ADC, if you need the maximum input voltage range (about 2.55V).

VojtechBartoska, 2022-03-30:
Hello @saeedbala, sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska