espressif/arduino-esp32 #5560: esp32 ble server - Method "onWrite" in characteristicCallBack not calling

AndrewPst, 2021-08-18:
I use esp32 4 MB of flash memory as a BLE server. I tried to make 2 services. The first service should send data about the battery charge level. The second one should communicate with the connected phone. After creating the services, esp32 stopped accepting incoming data. After deleting the battery service, the "onWrite" method was not called. I recreated the project, updated the platform, but the error remained. On another esp32 board, the code works. What happened and how to fix it?

Board:							ESP32 Dev Module
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

This code works on the new board, but does not work on the old one:
```
#include <Arduino.h>

#include <BLEServer.h>
#include <BLEService.h>
#include <BLEDevice.h>
#include <BLE2902.h>

#define SERVICE_UUID "C6FBDD3C-7123-4C9E-86AB-005F1A7EDA01" //при изменении не забудьте поменять и в приложении
#define CHARACTERISTIC_UUID "B88E098B-E464-4B54-B827-79EB2B150A9F"

BLEServer *server;
BLEService *service;
BLECharacteristic *characteristic;

BLEDescriptor *descriptor;

class ServerCallBack : public BLEServerCallbacks
{
  void onConnect(BLEServer *server) override
  {
    log_i("connect");
  }
  void onDisconnect(BLEServer *server) override
  {
    log_i("disconnect");
  }
};

class CharacteristicCallBack : public BLECharacteristicCallbacks
{
public:
  //This method not called
  void onWrite(BLECharacteristic *characteristic_) override
  {
    log_i("data is received"); 
  }
};

class DescrCallBack : public BLEDescriptorCallbacks
{
public:
  void onWrite(BLEDescriptor *desc) override
  {
    log_i("Data received!");
  }
};

void setup()
{
  Serial.begin(115200);

  BLEDevice::init("esp32_ble");
  BLEDevice::setPower(ESP_PWR_LVL_P3);
  server = BLEDevice::createServer();
  server->setCallbacks(new ServerCallBack());

  service = server->createService(SERVICE_UUID);
  characteristic = service->createCharacteristic(CHARACTERISTIC_UUID,
                                                 BLECharacteristic::PROPERTY_WRITE | BLECharacteristic::PROPERTY_WRITE_NR );

  descriptor = new BLE2902();
  descriptor->setCallbacks(new DescrCallBack());
  characteristic->addDescriptor(descriptor);
  characteristic->setCallbacks(new CharacteristicCallBack());
  service->start();
  const char *macAddress_ = BLEDevice::getAddress().toString().c_str();
  Serial.println(macAddress_);
  server->startAdvertising();
}

void loop()
{
  // put your main code here, to run repeatedly:
}

```

k-skor, 2021-10-14:
Actually my last post is not the actual solution. In order to debug it I've enabled logs on ESP32 and found out that the write message that comes is a chunked write. But the ending of the chunked message is never sent by my Android client. On nRF Connect mobile app there is option of a "Reliable Write" and then it works and "onWrite" method gets called.

No idea why it works with ESP-wroom-32D and desn't work with ESP32-wroom-32 though.

Also it works when you send message as text not byte but no idea why.

chegewara, 2021-10-14:
If i only have any idea what may be the cause i would write.
It snot that i know everything, or everything about this library, i just from experience know that sometimes very odd things may happen with the same code/binary on different boards. Especially if one is custom board and other devkit (im not saying its your case).
What i can recommend it to use nrf connect android app to perform test with both devices and then see if it works or not.

VojtechBartoska, 2022-03-30:
Up to last comment, I'm closing this issue as you found a solution. :)

closed 2022-03-30 by VojtechBartoska

arkonsolutions, 2025-04-07:
> For me it worked when I've added the second onWrite overload:
> 
> ```
>     void onWrite(BLECharacteristic *pCharacteristic, esp_ble_gatts_cb_param_t* param) {
>       Serial.println("Characteristic Write 1");
>       onWrite(pCharacteristic);
>     }
> ```

I spent 6 hours on this today and the solution turned out to be using this method. Thanks for the advice.