espressif/arduino-esp32 #5562: ESP32 Arduino Make error in IDF make project (IDF version 4.3)

Mos135, 2021-08-19:
### Hardware:
Board:							ESP32-WROOM-32E
Core Installation version:			Latest
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I startet the simple ("hello_world") project for Arduino to use in ESP-IDF as link:
https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html#esp32-arduino-lib-builder

I did as:
mkdir components
cd components
git clone https://github.com/espressif/arduino-esp32.git arduino 
cd arduino
git submodule update --init --recursive
cd ../.. 
idf.py menuconfig
On Autostart Arduino setup and loop on boot.
idf.py build

but I got the error as here:

Generated C:/Users/MNa/Projekt/hello/build/bootloader/bootloader.bin
[1002/1111] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/libb64/cdecode.c.obj
../components/arduino/cores/esp32/libb64/cdecode.c: In function 'base64_decode_block_signed':
../components/arduino/cores/esp32/libb64/cdecode.c:42:23: warning: this statement may fall through [-Wimplicit-fallthrough=]
         *plainchar    = (fragment & 0x03f) << 2;
         ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/cores/esp32/libb64/cdecode.c:43:7: note: here
       case step_b:
       ^~~~
../components/arduino/cores/esp32/libb64/cdecode.c:53:23: warning: this statement may fall through [-Wimplicit-fallthrough=]
         *plainchar    = (fragment & 0x00f) << 4;
         ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/cores/esp32/libb64/cdecode.c:54:7: note: here
       case step_c:
       ^~~~
../components/arduino/cores/esp32/libb64/cdecode.c:64:23: warning: this statement may fall through [-Wimplicit-fallthrough=]
         *plainchar    = (fragment & 0x003) << 6;
         ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/cores/esp32/libb64/cdecode.c:65:7: note: here
       case step_d:
       ^~~~
[1006/1111] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/libb64/cencode.c.obj
../components/arduino/cores/esp32/libb64/cencode.c: In function 'base64_encode_block':
../components/arduino/cores/esp32/libb64/cencode.c:46:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
             result = (fragment & 0x003) << 4;
             ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/cores/esp32/libb64/cencode.c:47:9: note: here
         case step_B:
         ^~~~
../components/arduino/cores/esp32/libb64/cencode.c:56:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
             result = (fragment & 0x00f) << 2;
             ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/cores/esp32/libb64/cencode.c:57:9: note: here
         case step_C:
         ^~~~
[1022/1111] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/firmware_msc_fat.c.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/firmware_msc_fat.c.obj

//I Change the code below and rename the sketch to main.cpp
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

me-no-dev, 2021-08-19:
```
[1022/1111] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/firmware_msc_fat.c.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/firmware_msc_fat.c.obj
```

This does not really show an error. Which chip are you building for?

Mos135, 2021-08-19:
> ```
> [1022/1111] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/firmware_msc_fat.c.obj
> FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/firmware_msc_fat.c.obj
> ```
> 
> This does not really show an error. Which chip are you building for?

thank you for the answer.
when I use "idf.py build" again it shows following error:

../components/arduino/cores/esp32/firmware_msc_fat.c:32:5: error: statement with no effect [-Werror=unused-value]
     for(l; l < max_len; l++){
     ^~~
cc1.exe: some warnings being treated as errors
[9/86] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WebServer/src/WebServer.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

I use ESP32-WROOM-32E

Mos135, 2021-08-19:
I have applied idf.py build many times and this is the final error that shows any time:
../components/arduino/cores/esp32/firmware_msc_fat.c:32:5: error: statement with no effect [-Werror=unused-value]
     for(l; l < max_len; l++){
     ^~~
cc1.exe: some warnings being treated as errors
[3/9] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFi/src/ETH.cpp.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFi/src/ETH.cpp.obj
.
.
../components/arduino/libraries/WiFi/src/ETH.cpp:291:23: error: 'esp_eth_phy_new_ksz8081' was not declared in this scope
             eth_phy = esp_eth_phy_new_ksz8081(&phy_config);
                       ^~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/libraries/WiFi/src/ETH.cpp:291:23: note: suggested alternative: 'esp_eth_phy_new_ksz8041'
             eth_phy = esp_eth_phy_new_ksz8081(&phy_config);
                       ^~~~~~~~~~~~~~~~~~~~~~~
                       esp_eth_phy_new_ksz8041
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

me-no-dev, 2021-08-19:
`error: statement with no effect` is fixed in master. For the other error, please use ESP-IDF master branch.

closed 2021-08-19 by me-no-dev