espressif/arduino-esp32 #5567: 2.0.0-rc1 sd_test example card mount failed

ismavg, 2021-08-19:
----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32S2 custom board (module selected is ESP32S2 dev module)
Core Installation version:			2.0.0-rc1
IDE name:						Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                      yes
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
I am trying to run the sd_test example on one of my custom boards for ESP32S2 . The PINs used for SPI are different from than the standard for which i used the SPIClass.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

/*
 * Connect the SD card to the following pins:
 *
 * SD Card | ESP32
 *    D2       -
 *    D3       SS
 *    CMD      MOSI
 *    VSS      GND
 *    VDD      3.3V
 *    CLK      SCK
 *    VSS      GND
 *    D0       MISO
 *    D1       -
 */
#include "FS.h"
#include "SD.h"
#include "SPI.h"
#define miso 40
#define mosi 38
#define sck  39
#define ss   37

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void createDir(fs::FS &fs, const char * path){
    Serial.printf("Creating Dir: %s\n", path);
    if(fs.mkdir(path)){
        Serial.println("Dir created");
    } else {
        Serial.println("mkdir failed");
    }
}

void removeDir(fs::FS &fs, const char * path){
    Serial.printf("Removing Dir: %s\n", path);
    if(fs.rmdir(path)){
        Serial.println("Dir removed");
    } else {
        Serial.println("rmdir failed");
    }
}

void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\n", path);

    File file = fs.open(path);
    if(!file){
        Serial.println("Failed to open file for reading");
        return;
    }

    Serial.print("Read from file: ");
    while(file.available()){
        Serial.write(file.read());
    }
    file.close();
}

void writeFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Writing file: %s\n", path);

    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }
    if(file.print(message)){
        Serial.println("File written");
    } else {
        Serial.println("Write failed");
    }
    file.close();
}

void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("Failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("Message appended");
    } else {
        Serial.println("Append failed");
    }
    file.close();
}

void renameFile(fs::FS &fs, const char * path1, const char * path2){
    Serial.printf("Renaming file %s to %s\n", path1, path2);
    if (fs.rename(path1, path2)) {
        Serial.println("File renamed");
    } else {
        Serial.println("Rename failed");
    }
}

void deleteFile(fs::FS &fs, const char * path){
    Serial.printf("Deleting file: %s\n", path);
    if(fs.remove(path)){
        Serial.println("File deleted");
    } else {
        Serial.println("Delete failed");
    }
}

void testFileIO(fs::FS &fs, const char * path){
    File file = fs.open(path);
    static uint8_t buf[512];
    size_t len = 0;
    uint32_t start = millis();
    uint32_t end = start;
    if(file){
        len = file.size();
        size_t flen = len;
        start = millis();
        while(len){
            size_t toRead = len;
            if(toRead > 512){
                toRead = 512;
            }
            file.read(buf, toRead);
            len -= toRead;
        }
        end = millis() - start;
        Serial.printf("%u bytes read for %u ms\n", flen, end);
        file.close();
    } else {
        Serial.println("Failed to open file for reading");
    }

    file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }

    size_t i;
    start = millis();
    for(i=0; i<2048; i++){
        file.write(buf, 512);
    }
    end = millis() - start;
    Serial.printf("%u bytes written for %u ms\n", 2048 * 512, end);
    file.close();
}

void setup(){
    Serial.begin(115200);
    static SPIClass* spi = NULL;
    spi = new SPIClass(2);
    spi->begin(sck, miso, mosi, ss);
    if(!SD.begin(ss, *spi, 40000000)){
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD card attached");
        return;
    }

    Serial.print("SD Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);

    listDir(SD, "/", 0);
    createDir(SD, "/mydir");
    listDir(SD, "/", 0);
    removeDir(SD, "/mydir");
    listDir(SD, "/", 2);
    writeFile(SD, "/hello.txt", "Hello ");
    appendFile(SD, "/hello.txt", "World!\n");
    readFile(SD, "/hello.txt");
    deleteFile(SD, "/foo.txt");
    renameFile(SD, "/hello.txt", "/foo.txt");
    readFile(SD, "/foo.txt");
    testFileIO(SD, "/test.txt");
    Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
}

void loop(){

}
```

### Debug Messages:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x40c
load:0x4004c000,len:0x850
load:0x40050000,len:0x248c
SHA-256 comparison failed:
Calculated: 791605df05071c12acbff4f397d899cbafb72e37cbb68bf21d8f2c8a8de12ab1
Expected: f733c5b73caafbbdfdbf084499e97098c114de0fffc146a6fd6fc50138d3038c
Attempting to boot anyway...
entry 0x4004c17c
[  1126][D][esp32-hal-psram.c:84] psramInit(): PSRAM enabled
⸮[  1134][W][sd_diskio.cpp:180] sdCommand(): crc error
[  1235][W][sd_diskio.cpp:180] sdCommand(): crc error
[  1335][W][sd_diskio.cpp:180] sdCommand(): crc error
[  1435][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[  1435][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1440][W][sd_diskio.cpp:180] sdCommand(): crc error
[  1544][W][sd_diskio.cpp:180] sdCommand(): crc error
[  1644][W][sd_diskio.cpp:180] sdCommand(): crc error
Card Mount Failed
```

Jason2866, 2021-08-19:
Maybe the same reason why #5550 fails too.

Lawrence-RedFern, 2021-09-16:
Did you ever come to a resolution for this? I am having the [exact same issue](https://github.com/espressif/arduino-esp32/issues/5676).

ismavg, 2021-09-17:
no i have not found any solution yet

ismavg, 2021-11-08:
Hi @P-R-O-C-H-Y,
thank you for the response.
But, as per my knowledge, [we change the pin number ](https://github.com/espressif/arduino-esp32/issues/5676)
correct me if I'm wrong

P-R-O-C-H-Y, 2021-11-08:
Hi, yes the esp32s2 use GPIO-matrix so you can use almost every pin you want to.
So I tested it again with exact your code and pin use, and its working perfectly. Idk why its not working for you.
Can you try to update arduino-esp32 to latest version?
Just an advice, the best way is to use predefined pins in data sheet which I linked before. If you use matrix you may not be able to get best pin speeds out of SPI.

ismavg, 2021-11-09:
I have tried the latest version. and also the same pins are working in IDF
My defines mean GPIO pin number.

P-R-O-C-H-Y, 2021-11-09:
It is working for me. With your PINS in IDF and in Arduino.
Does the SD module you are using needs 3.3V or 5V power?
So in ESP-IDF its working for you?

ismavg, 2021-11-10:
yes, idf works fine.

P-R-O-C-H-Y, 2021-11-15:
I think it might be more HW related. For me it is working in Arduino.
Does your model needs 3.3V or 5V power? What about pull-ups resistors on used pins connected to SD card?
What module are you using?

Sample code runs fine.
```
#include <SD.h>

#define SD_MOSI  38
#define SD_MISO  40
#define SD_SCK   39
#define SD_CS    37

void setup() {

  Serial.begin(115200);
  
  SPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS); 

  if (!SD.begin(SD_CS)) {
    Serial.println("initialization failed!");
    while (1);
  }
  Serial.println("initialization done.");
}

void loop() {
  // your code here
}
```

ismavg, 2021-11-16:
It is a custom board with an esp32 and sd card slot onboard. 3.3V converter on board. everything work fine in idf so i don't think it is a hardware issue.

P-R-O-C-H-Y, 2021-12-14:
I can't reproduce your problem, because you have a custom board. On Espressif's boards everything is working fine.

closed 2021-12-14 by P-R-O-C-H-Y