espressif/arduino-esp32 #5572: Upload fails for Sparkfun ESP32 Thing Plus {upload.flags}

riosnav, 2021-08-21:
### Hardware:
Board:							SparkFun ESP32 Thing Plus (WRL-17381)
Core Installation version:			2.0.0-rc1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     No?
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
When trying to upload a sketch using "SparkFun ESP32 Thing Plus" as the board in the Arduino IDE I get the following error message:

```
Sketch uses 760949 bytes (11%) of program storage space. Maximum is 6553600 bytes.
Global variables use 34604 bytes (10%) of dynamic memory, leaving 293076 bytes for local variables. Maximum is 327680 bytes.
C:\Users\riosnav\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0/esptool.exe --chip esp32 --port COM4 --baud 921600 {upload.flags} --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\riosnav\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/partitions/boot_app0.bin 0x1000 C:\Users\riosnav\AppData\Local\Temp\arduino_build_44862/ESP32_weather_station.ino.bootloader.bin 0x10000 C:\Users\riosnav\AppData\Local\Temp\arduino_build_44862/ESP32_weather_station.ino.bin 0x8000 C:\Users\riosnav\AppData\Local\Temp\arduino_build_44862/ESP32_weather_station.ino.partitions.bin 
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}]
               [--port PORT] [--baud BAUD]
               [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset,no_reset_stub}]
               [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
               ...
esptool: error: argument operation: invalid choice: '{upload.flags}' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'merge_bin', 'version', 'get_security_info')
esptool: error: argument operation: invalid choice: '{upload.flags}' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'merge_bin', 'version', 'get_security_info')
```

Using "Adafruit ESP32 Feather" as stated in the [ESP32 Thing Plus Hookup Guide by Sparkfun](https://learn.sparkfun.com/tutorials/esp32-thing-plus-hookup-guide#software-setup) (it was done before the board was added to the list) works without an issue:
```
Sketch uses 760949 bytes (58%) of program storage space. Maximum is 1310720 bytes.
Global variables use 34604 bytes (10%) of dynamic memory, leaving 293076 bytes for local variables. Maximum is 327680 bytes.
C:\Users\riosnav\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0/esptool.exe --chip esp32 --port COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\riosnav\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/partitions/boot_app0.bin 0x1000 C:\Users\riosnav\AppData\Local\Temp\arduino_build_44862/ESP32_weather_station.ino.bootloader.bin 0x10000 C:\Users\riosnav\AppData\Local\Temp\arduino_build_44862/ESP32_weather_station.ino.bin 0x8000 C:\Users\riosnav\AppData\Local\Temp\arduino_build_44862/ESP32_weather_station.ino.partitions.bin 
esptool.py v3.1
Serial port COM4
Connecting......
```

Comparing both commands I would guess that {upload.flags} shouldn't be there.

Also there is a huge difference in program storage space between both boards (6553600 vs 1310720 bytes), is this correct? It seems odd to me considering that the manufacturer appears to consider them equivalent.

Thank you.

riosnav, 2021-08-23:
I just tried to upload the sketch after updating to RC2 and the issue is still present.

atanisoft, 2021-08-23:
@riosnav Update the boards.txt file to add:
```
esp32thing_plus.upload.flags=
```
Which appears to have been missed for this board, possibly others.

riosnav, 2021-08-23:
> @riosnav Update the boards.txt file to add:
> 
> ```
> esp32thing_plus.upload.flags=
> ```
> 
> Which appears to have been missed for this board, possibly others.

This did the trick! Thank you.

PR #5589 (2021-08-25): Add missing upload.flags on boards.txt
## Summary
Fixes #5572.

There were several boards that were missing the _upload.flags_ and _upload.extra_flags_ fields, resulting in an error during upload of a sketch.
closed 2021-08-25 by me-no-dev