espressif/arduino-esp32 #5575: ESP32-C3 RISC-V mtime CPU timer register address declaration missing

helmut64, 2021-08-22:
Has someone an idea where the address of the mtime and mtimecmp RISC-V address is mapped in the C3 address space?
I like to use the CPU build in 64-bit timer register which is standard in the RISC-V spec.

lbernstone, 2021-08-22:
arduino-esp32 is a HAL based on a framework, so I don't think this is the correct forum for this q.  The technical reference manual is available at espressif.com
[image]

helmut64, 2021-08-22:
I am using the C3 with Arduino, that is why I asked here. I know the reference manual however the RISC-V mtime address is not listed there. I was hopping that the mtime feature gives me accurate timestamps to avoid an extra timer.

igrr, 2021-08-22:
Hi @helmut64, the CPU in ESP32-C3 indeed doesn't have an mtime CSR.
If you only need the timestamps, you can call `cpu_hal_get_cycle_count` function — it relies on the fact that at startup time, the performance counter CSR is initialized to count CPU cycles. So the corresponding counter CSR can be used as a cycle counter.
If you also need some kind of interrupt functionality, you can use the memory-mapped Timer Group peripheral.

helmut64, 2021-08-23:
Hi Ivan, I found a solution using the esp_timer_get_time() function which seems to use a microsecond resolution and returns an int64 value which means no overflow problems. It also handles sleep cases and adjusts it’s values.
On the ESP32-Xtensa I used two of the four timers, on the C3 I will use the esp_timer_get_time() etc. functions to be more generic and to avoid to occupy timers because the C3 has less of them.
BTW: The Arduino esp32-hal-timer.c/.h code has not been adjusted for the C3 available timers.

Regards Helmut
I close this case because I know everything I need.

closed 2021-08-23 by helmut64