espressif/arduino-esp32 #5577: WiFi.mode(WIFI_OFF) prevents future mode changes

FromenActual, 2021-08-22:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			2.0.0-rc1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
Had an old piece of code on my desktop that toggles the WiFi mode between STA and OFF, which used to work fine. Got a new laptop and set it up for programming the ESP32, at which point the same code no longer works. Specifically, once `WiFi.mode(WIFI_OFF)` is called, `WiFi.mode(WIFI_STA)` no longer changes the mode. It appears there's some initialization error occuring when it tries to enable STA mode.

My desktop has the 1.0.4 version of the board manager installed, whereas my laptop has 2.0.0-rc-1 installed (latest as of posting this). The code below is a simplified version that illustrates the issue:

### Sketch:
```cpp
#include <WiFi.h>

void setup() {
  // Init Serial Monitor
  Serial.begin(115200);
  delay(100); // Give a bit of time for Serial to become available
 
  Serial.print("Mode undefined, probably 0 for OFF mode:");
  Serial.println(WiFi.getMode());
  delay(1000);
}
 
void loop()
{
  // Set STA mode
  WiFi.mode(WIFI_STA);
  Serial.print("STA mode, should be 1:");
  Serial.println(WiFi.getMode());
  delay(1000);

  // Set OFF mode
  WiFi.mode(WIFI_OFF);
  Serial.print("OFF mode, should be 0:");
  Serial.println(WiFi.getMode());
  delay(1000);
}
```

### Debug Messages:
Serial output on my desktop with Core debug level: Debug:
```
Mode undefined, probably 0 for OFF mode:0
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
STA mode, should be 1:1
OFF mode, should be 0:0
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
STA mode, should be 1:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
OFF mode, should be 0:0
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
STA mode, should be 1:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
OFF mode, should be 0:0
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
```

Serial output on my laptop with Core debug level: Debug:
```
Mode undefined, probably 0 for OFF mode:0
[  1119][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 0 - WIFI_READY
STA mode, should be 1:1
[  1207][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 2 - STA_START
[  2207][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 3 - STA_STOP
OFF mode, should be 0:0
[  3209][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
STA mode, should be 1:0
OFF mode, should be 0:0
[  5209][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
STA mode, should be 1:0
OFF mode, should be 0:0
[  7209][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
```

`12289 = ESP_ERR_WIFI_NOT_INIT`, which to me sounds like `WiFi.mode(WIFI_STA)` is perhaps missing something? Or is there a step I've missed in my code somewhere? Haven't found many sources on how to re-enable STA mode, so my code is a bit of a guess. Regardless, I'll revert my laptop to 1.0.4 for now, which I can confirm fixes the problem.

qnxsgwy, 2021-08-25:
I seem to have this problem too, and hope it can be solved.

Villaquiranm, 2021-08-30:
Duplicate of #4842

Villaquiranm, 2021-08-30:
Hello this seems like a duplicate issue it seems to be already fixed and merged to the master branch :)

FromenActual, 2021-08-30:
You're right, thanks for letting me know! Glad this got resolved quickly 😃

closed 2021-08-30 by FromenActual