espressif/arduino-esp32 #5578: r.2.0.0,I2S meet the mistakes as follow:

xpdrycry, 2021-08-23:
> Executing task: C:\Users\Administrator.1-PC\.platformio\penv\Scripts\platformio.exe run <

Processing esp32c3 (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream; framework: arduino; board: esp32-c3-devkitm-1)
-------------------------------------------------------------------------------------------------------------------------
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r1
Verbose mode can be enabled via `-v, --verbose` option
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r1
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h,
olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.4ada3f5
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-riscv-esp 1.80400.0 (8.4.0)
 - toolchain-riscv32-esp 8.4.0+2021r1
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <audio_processor>
|   |-- <audio_input>
|-- <audio_input>
|-- <neural_network>
|   |-- <tfmicro>
|-- <DNSServer> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <WiFi> 2.0.0
|-- <EEPROM> 2.0.0
Building in release mode
Linking .pio\build\esp32c3\firmware.elf
c:/users/administrator.1-pc/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/Administrator.1-PC/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/lib\libdriver.a(i2s.c.obj): in function `i2s_set_clk':
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/i2s.c:390: undefined reference to `rtc_clk_apll_enable'
c:/users/administrator.1-pc/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/Administrator.1-PC/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/lib\libhal.a(i2s_hal.c.obj): in function `i2s_hal_init':
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/i2s_hal.c:273: undefined reference to `I2S0'
c:/users/administrator.1-pc/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/i2s_hal.c:273: undefined reference to `I2S0'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32c3\firmware.elf] Error 1

atanisoft, 2021-08-23:
Duplicate of https://github.com/espressif/esp-idf/issues/7114.

VojtechBartoska, 2022-03-30:
Duplicate of #5597, I'm closing this one. @xpdrycry please continue discussion in mentioned issue so we don't have 3 same issues. Thanks!

closed 2022-03-30 by VojtechBartoska