espressif/arduino-esp32 #5580: 2.0.0-rc2 - Serial.setRxBufferSize() is missing

szerwi, 2021-08-23:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			2.0.0-rc2
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Serial.setRxBufferSize() function is missing in HardwareSerial library in 2.0.0-rc2.

SuGlider, 2021-08-23:
@szerwi buffer size can be set using begin(...) Function.

It is one of the parameters.
Default RX buffer size is 256 bytes.

szerwi, 2021-08-23:
@SuGlider how can I set it in the `begin()` function?
From HardwareSerial.h: `void begin(unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false, unsigned long timeout_ms = 20000UL, uint8_t rxfifo_full_thrhd = 112);`
Is that a `rxfifo_full_thrhd` parameter?

atanisoft, 2021-08-23:
@szerwi It's not currently exposed in HardwareSerial. @SuGlider can you add another parameter to expose it?

SuGlider, 2021-08-23:
@szerwi, @atanisoft parameter exposed in PR #5583

SuGlider, 2021-08-23:
HardwareSerial::setRxBufferSize(size_t newSize) has been added and commited.

PR #5583 (2021-08-24): Adds HardwareSerial::setRxBufferSize()
## Summary
It adds HardwareSerial::setRxBufferSize() with default value of 256 bytes.
Fixes #5580 

## Impact
It allows to define a Rx Buffer Size for UART, before begin().
closed 2021-08-24 by me-no-dev