espressif/arduino-esp32 #5584: WiFiClientSecure not working lwip mode ! (insecure)

elexal, 2021-08-23:
hello friend
we have an ESP8266 that connects to the host SSL with wifi 
And we receive data from it without any problems  
We have an interface Ethernet(W5500 or Enc28j60) which we want to receive data from it
If the SSL does not exist it connects to the server and receives the data with wificlient and lwip library
When we activate Setinsecure and WiFiClientSecure  it doesn’t connect to the server anymore.  
The server that we want to connect to by Ethernet has SSL and we have to use wificlient secure library that works without any problem on wifi , but on Ethernet mode it doesn’t connect
Codes for wifi and Ethernet are the same and are used from one library
In the following example we connect to a host by Ethernet and receive data but if we change the library from wifi client to wificlient secure it doesn’t work anymore. 
`/*
    This sketch establishes a TCP connection to a "quote of the day" service.
    It sends a "hello" message, and then prints received data.
*/

#include <SPI.h>
#include <ENC28J60lwIP.h>

#include <ESP8266WiFi.h>

const char* host = "djxmmx.net";
const uint16_t port = 17;

#define CSPIN D1

ENC28J60lwIP eth(CSPIN);

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

IPAddress server(23, 28, 179, 206);

void setup() {
  Serial.begin(115200);
  delay(500);

  WiFi.mode(WIFI_OFF);

  SPI.begin();
//  if (!eth.config(ip, gw, nm, gw, IPADDR_NONE)) {
//    Serial.println("wrong config");
//  }
  eth.setDefault(); // use ethernet for default route
  if (!eth.begin(mac)) {
    Serial.println("ethernet hardware not found ... sleeping");
    while (1) {
      delay(1000);
    }
  } else {
    Serial.print("connecting ethernet");
    while (!eth.connected()) {
      Serial.print(".");
      delay(1000);
    }
  }
  Serial.println();
  Serial.print("ethernet IP address: ");
  Serial.println(eth.localIP());
  Serial.println(eth.gatewayIP());
  Serial.println(eth.subnetMask());
  Serial.println(WiFi.dnsIP());
}

void loop() {
  static bool wait = true;

  Serial.print("connecting to ");
  Serial.print(host);
  Serial.print(':');
  Serial.println(port);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    delay(5000);
    return;
  }

  // This will send a string to the server
  Serial.println("sending data to server");
  if (client.connected()) {
    client.println("hello from ESP8266");
  }

  // wait for data to be available
  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 5000) {
      Serial.println(">>> Client Timeout !");
      client.stop();
      delay(60000);
      return;
    }
  }

  // Read all the lines of the reply from server and print them to Serial
  Serial.println("receiving from remote server");
  // not testing 'client.connected()' since we do not need to send data here
  while (client.available()) {
    char ch = static_cast<char>(client.read());
    Serial.print(ch);
  }

  // Close the connection
  Serial.println();
  Serial.println("closing connection");
  client.stop();

  if (wait) {
    delay(300000); // execute once every 5 minutes, don't flood remote service
  }
  wait = true;
}`

atanisoft, 2021-08-23:
@elexal Please file this against https://github.com/esp8266/Arduino as this repository is only for ESP32.

VojtechBartoska, 2022-03-30:
Hello, closing. Thanks @atanisoft for your reply.

closed 2022-03-30 by VojtechBartoska