espressif/arduino-esp32 #5587: RepeatTimer and WatchdogTimer both not work

Bei-Ji-Quan, 2021-08-24:
Hi
I use the two examples RepeatTimer and WatchdogTimer with ESP32-S2-Saola-1. RepeatTimer  print nothing , and WatchdogTimer  does not reset when i pressed down the boot key for a long time.
I also tested the two example with Wifi-lora-32-V2(produce by Heltec Automation), It's the same result. But Wifi-lora-32-V2 works with old Arduino code.

VojtechBartoska, 2022-03-30:
Hello @Bei-Ji-Quan, sorry for late reply. Timer was refactored to use ESP-IDF API. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska