espressif/arduino-esp32 #5588: spiffs 3MB upload failed

minhhieuec, 2021-08-24:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.6
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:				        921600
Computer OS:                                         Windows 10

### Description:
i'm using appendFile function in SPIFFS_Test.ino file to test spiffs of esp32. But maximum size of characters i can write to the file is 2240640 bytes (not match with 3MB SPIFFS define in partition file).
```
- Select flash size 4MB
- Select partition scheme: No OTA (1MB APP/3MB SPIFFS)
```

minhhieuec, 2021-08-24:
[dat1.txt](https://github.com/espressif/arduino-esp32/files/7038899/dat1.txt)

VojtechBartoska, 2022-03-30:
Hello @minhhieuec , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-14:
Hello,

as there was no answer in last 14 days, I'm closing the issue as expired to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-14 by VojtechBartoska