espressif/arduino-esp32 #5591: fs.flush() method from FS library doesn't do anything

sw-dev-code, 2021-08-25:
I would like to do the file flush when my file is over the size limit. When I call file.flush() method on my file nothing happens. 

I'm using the ESP32-DevKitC and PlatformIO development environment.

atanisoft, 2021-08-25:
FS.flush() writes data to the filesystem that hasn't already been synced to storage (SD, SPIFFS, LittleFS, etc). It will not shrink/resize the file.

There is no Arduino FS API for truncating the file. The easiest approach would be to call `File.close()`, `FS.remove("file")` and then recreate the file.

sw-dev-code, 2021-08-25:
@atanisoft Ok, sorry, I thought that flush will erase the content of a file.

I have file limitation, so when it reach 100kB I need to append only the last data.

Thank you for your help. Closing the issue.

closed 2021-08-25 by sw-dev-code

atanisoft, 2021-08-25:
@sw-dev-code What you are looking for is `truncate()` but that is not part of the Arduino FS spec. There are a few posts out there for adding it but doesn't look like it has gained any traction.