espressif/arduino-esp32 #5604: ESP32-S2/C3 build of 2.0.0-rc2 fails

domenk, 2021-08-29:
When trying to use [esp32-arduino-lib-builder](https://github.com/espressif/esp32-arduino-lib-builder), ESP32-S2 and ESP32-C3 builds fail with the error below. ESP32 builds fine.

Error:
<pre>
../components/arduino/cores/esp32/esp32-hal-timer.c: In function '__timerISR':
../components/arduino/cores/esp32/esp32-hal-timer.c:99:26: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_st'
     uint32_t s0 = TIMERG0.int_st.val;
                          ^
../components/arduino/cores/esp32/esp32-hal-timer.c:100:26: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_st'
     uint32_t s1 = TIMERG1.int_st.val;
                          ^
../components/arduino/cores/esp32/esp32-hal-timer.c:101:12: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_clr'
     TIMERG0.int_clr.val = s0;
            ^
../components/arduino/cores/esp32/esp32-hal-timer.c:102:12: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_clr'
     TIMERG1.int_clr.val = s1;
            ^
../components/arduino/cores/esp32/esp32-hal-timer.c: In function 'timerBegin':
../components/arduino/cores/esp32/esp32-hal-timer.c:241:16: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_ena'
         TIMERG1.int_ena.val &= ~BIT(timer->timer);
                ^
../components/arduino/cores/esp32/esp32-hal-timer.c:245:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_clr'
             TIMERG1.int_clr.val = BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:248:16: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_ena'
         TIMERG0.int_ena.val &= ~BIT(timer->timer);
                ^
../components/arduino/cores/esp32/esp32-hal-timer.c:252:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_clr'
             TIMERG0.int_clr.val = BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c: In function 'timerAttachInterrupt':
../components/arduino/cores/esp32/esp32-hal-timer.c:291:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_ena'
             TIMERG1.int_ena.val &= ~BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:295:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_clr'
             TIMERG1.int_clr.val = BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:298:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_ena'
             TIMERG0.int_ena.val &= ~BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:302:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_clr'
             TIMERG0.int_clr.val = BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:335:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_ena'
             TIMERG1.int_ena.val |= BIT(timer->timer);
                    ^
../components/arduino/cores/esp32/esp32-hal-timer.c:337:20: error: 'timg_dev_t' {aka 'struct <anonymous>'} has no member named 'int_ena'
             TIMERG0.int_ena.val |= BIT(timer->timer);
                    ^
[1243/1359] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-time.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
</pre>

me-no-dev, 2021-08-30:
this is because of recent change in ESP-IDF. Will be fixed

me-no-dev, 2021-08-31:
it's fixed in the `idf-master` branch of this repo :)

closed 2021-08-31 by me-no-dev

domenk, 2021-08-31:
For anyone still dealing with this issue: with this update, esp32-arduino-lib-builder successfully finishes building custom library for ESP32-S2/C3, but now Arduino sketch build fails with a similar error. To fix this, update files from folder [arduino-esp32/cores/esp32/](https://github.com/espressif/arduino-esp32/tree/idf-master/cores/esp32) (on branch `idf-master`) in your Arduino board installation (location on Windows: `%USERPROFILE%\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32\`).