espressif/arduino-esp32 #5607: Equivalent to MDNS.announce()

tk97tk, 2021-08-29:
Is there any equivalent method to MDNS.announce() from the ESP8266mDNS.h in the <ESPmDNS.h> library?
To my understanding the MDNS.announce() triggers a new MDNS announcement, this will lead to a call to 'MDNSDynamicServiceTxtCallback', which will update the time TXT item.

ESP8266mDNS.h:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/src/ESP8266mDNS.h

LEAmDNS.h (in which announce() is declared):
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/src/LEAmDNS.h

hreintke, 2021-08-29:
The MDNS.announce() only triggers the MDNS announcement.

When you add an explicit call f.e `MDNS.addDynamicServiceTxt(p_hService, "curtime", getTimeString());` in your code it will also get a  `MDNSDynamicServiceTxtCallback`.
But that is then also what you requested.

tk97tk, 2021-08-29:
Ok, let's forget about that MDNSDynamicServiceTxtCallback :D. 
How can I only trigger MDNS announcement with  <ESPmDNS.h>, so we have an equivalent to the MDNS.announce() from  ESP8266mDNS.h. Sorry for the misleading, thanks for your fast response :)

me-no-dev, 2021-08-30:
There is no need to implement `announce()` for ESP32. The MDNS stack is fully automated. It announces when there are interface changes (connect/change IP, etc.) and when there are new services added/changed/removed. All is done in the background for you.

closed 2021-08-30 by me-no-dev