espressif/arduino-esp32 #5608: ESP32-C3 Native USB missing in Arduino

helmut64, 2021-08-29:
I tested the native USB support with the Arduino "ESP32-S2 native USB" Board setting, this works perfect, also the USB drive is cool. Thank you, thank you, thank you.
Please add the same Arduino USB Board setting for the "ESP32-C3" Board. It has the same USB DFU/Serial features therefore it should be added. I even believe nothing/little needs to be programmed for this.

BTW: A Arduino common define for the boards,  independent if it is a USB DFU version or a UART Serial version would be helpful, e.g.:
-DESP32S2 
-DESP32C3
This would allow easy code differentiation based on the MCU, independent what kind of Serial/USB connection is being used.
Regards Helmut

me-no-dev, 2021-08-30:
C3's USB is nothing like S2. It is all in hardware and all we can access is the CDC(Serial) port. This will be added soon. You can not use the USB stack that is available for S2 on C3.

closed 2021-08-30 by me-no-dev

helmut64, 2021-08-30:
Thank you.
However I believe the C3 has the same USB DFU firmware upload like the S2, therefore it may makes sense to add this to be able to upload software.
USB CDC for the Arduino Serial Console would be another task.

me-no-dev, 2021-08-30:
I'll give it a shot :)

helmut64, 2021-08-30:
I am available for testing, we developed a couple of ESP32 commercial boards, see www.radioshuttle.de
Regards Helmut

me-no-dev, 2021-08-31:
released and part of 2.0.0 @helmut64

helmut64, 2021-08-31:
very cool, looked already into it, will test this tonight.

helmut64, 2021-08-31:
This is super great stuff, native USB works like a charm, I tested:
- S2 with native USB board target
- C3 with CDC on Boot enabled.
The Sketch-Upload as well the USB-CDC via the Serial Monitor works fine.
Thank you so much:
Warm regards, Helmut

s-light, 2022-01-10:
@helmut64 does it work reliable for you?
or / and is there anything special to do to get serial-output to the USB_CDC device from a arduino sketch reliable?
have you tried this from linux?

iam currently at [2.0.2](https://github.com/espressif/arduino-esp32/releases/tag/2.0.2)
and running arduino ide 1.8.19 on kubuntu 21.04

i have written all my tries in details [in this forum post](https://esp32.com/viewtopic.php?f=19&t=24280)
in short:
i now know that if `USB CDC on Boot:"Enabled"` i should just use `Serial` 
and if i have `USB CDC on Boot:"Disabled"` i have to use `USBSerial` to output to the USB_CDC.

upload of code works fine over the USB_CDC port¹.
and in both ways i often get no reliable output.
and if i only have USB_CDC connected i can not get it to output anything - 
if i power it from another source i sometimes get output when i replug the usb....

¹ if i have no output the auto-reset function also does not work. i have to manually enter `download mode`

iam looking forward to any tips :-)

s-light, 2022-06-21:
seems for me the last board-package update fixed things :-)
now it is working as expected.
if i set Tools → `USB CDC on Boot:"Enabled"` and use `Serial.println("Hello World")` i get `Hello World` on the CDC device :-)