espressif/arduino-esp32 #5610: compile error for SPIFFS.info(fs_info)

societyofrobots, 2021-08-29:
I'd like to print out SPIFFS file system info for ESP32, Arduino IDE, 1.0.6.

On [this webpage from 2018](https://circuits4you.com/2018/01/31/example-of-esp8266-flash-file-system-spiffs/), they use this code:
```
FSInfo fs_info;
SPIFFS.info(fs_info);
```

But it returns this error:
`'FSInfo' was not declared in this scope`

I only see fsinfo in the libraries and not fs_info, so I then tried:
`SPIFFS.info(fsinfo);`

But then I get this error:
`'class fs::SPIFFSFS' has no member named 'info'`

Was this removed, or changed to something else?

P-R-O-C-H-Y, 2021-10-12:
Hi @societyofrobots,

SPIFFS.info is not availible on ESP32, it was on the older ESP8266.

On ESP32 you can call these 2 functions to get info from the SPIFFS.
```
SPIFFS.totalBytes();
SPIFFS.usedBytes();
```

VojtechBartoska, 2021-12-14:
@societyofrobots Hello, can I consider this issue solved?

societyofrobots, 2021-12-14:
> @societyofrobots Hello, can I consider this issue solved?

P-R-O-C-H-Y 's solution worked.

closed 2021-12-14 by societyofrobots