espressif/arduino-esp32 #5611: how  to use esp32 PSRAM  ??

savagerex, 2021-08-30:
how to use esp32, psram.

do you have sample code???

----------------------------- Remove above -----------------------------

Hardware:
Hardware:
Board : ESP32 dev module
Core install version: 1.0.4
IDE name : Arduino
Flash Frequency : 80MHz
PSRAM Enable : yes
upload speed :115200
computer os : win10

lbernstone, 2021-08-30:
Use it for what purpose?
https://github.com/lbernstone/rrdtool_ESP32/blob/master/examples/psramDB/psramDB.ino

savagerex, 2021-08-30:
I have data and save to SPIFFS.

then read date from SPIFFS.

data will fill to const char *data.

it need ram to save

lbernstone, 2021-08-30:
In arduino-esp32 2.0, psram will automagically be used for large allocations (16k on esp32, 4k on new variants).  If you want to allocate smaller strings to psram, you will need to use ps_malloc to define the size of the variable (eg `char *data = ps_malloc(4096*sizeof(char));`).  Since I don't think I have posted it elsewhere, I'm also going to put up the code I have for a custom allocator for stdlib objects.

```
template <class T>
struct PSallocator {
  typedef T value_type;
  PSallocator() = default;
  template <class U> constexpr PSallocator(const PSallocator<U>&) noexcept {}
  [[nodiscard]] T* allocate(std::size_t n) {
    if(n > std::size_t(-1) / sizeof(T)) throw std::bad_alloc();
    if(auto p = static_cast<T*>(ps_malloc(n*sizeof(T)))) return p;
    throw std::bad_alloc();
  }
  void deallocate(T* p, std::size_t) noexcept { std::free(p); }
};
template <class T, class U>
bool operator==(const PSallocator<T>&, const PSallocator<U>&) { return true; }
template <class T, class U>
bool operator!=(const PSallocator<T>&, const PSallocator<U>&) { return false; }

std::vector<int, PSallocator<int> > v;

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println(ESP.getFreePsram());
  v.reserve(900000);
  for (uint32_t x=0; x<900000; x++) {
    v.push_back(x);
  }
  Serial.println(ESP.getFreePsram());
  v.clear();
  v.shrink_to_fit();
  Serial.println(ESP.getFreePsram());
}

void loop() {
  vTaskDelete(NULL);
}
```

savagerex, 2021-08-31:
OK, THX

savagerex, 2021-09-01:
I use psram in global.
esp32 always restart!!!

char* TempCaTotalData = (char*)ps_calloc(1024 , sizeof(char));

//char TempCaTotalData[10]={0};
void setup() {
  // put your setup code here, to run once:
   Serial.begin(115200);
   char* receivedDataOne = "123";
   char* receivedDataTwo = "456";
   char* receivedDataThree = "789";

  // TempCaTotalData = (char*)ps_calloc(1024 , sizeof(char));

   char* TempCaData = (char*)ps_calloc(128 , sizeof(char));
   TempCaData = receivedDataOne;
   strcat(TempCaTotalData, TempCaData);

   TempCaData = receivedDataTwo;
   strcat(TempCaTotalData, TempCaData);

   TempCaData = receivedDataThree;
   strcat(TempCaTotalData, TempCaData);

   Serial.print(TempCaTotalData);

   
}

chegewara, 2021-09-01:
Did you enable PSRAM in menu`tools->PSRAM`?

savagerex, 2021-09-01:
Yes.

you can try this code.

savagerex, 2021-09-01:
[image]

me-no-dev, 2021-09-01:
you CAN NOT call ps_xalloc in the global like that. PSRAM is not available at that point.
```cpp
char* TempCaTotalData = NULL;
void setup() {
    TempCaTotalData = (char*)ps_calloc(1024 , sizeof(char));
    // rest of the sketch
}
```

closed 2021-09-01 by me-no-dev