espressif/arduino-esp32 #5612: Encryption with Arduino IDE- and ESP-idf

Mos135, 2021-08-30:
Hi
I have written a complicated project with Arduino-IDE for ESP32, GPS and Display modules and it was working perfect.
After that I have decided to enable encryption and protection, to protect my program from copy. but I find out that it is not possible to enable it through Arduino-IDE. I read many forums and have tried to use this to using ESP-idf:

https://github.com/espressif/arduino-esp32/issues/1387

but the problem was, I got many errors from other module's libraries while I didn't have the same problem with Arduino-IDE.
then I have tried to compile my code with Arduino-IDE and only write the encrypted flash in the app partition and I got the error that it can not initialize EEPROM and SPIFF.
then I have tried to find out what the problem is. I write a simple code to use SPIFF and EEPROM. when I compile it in ESP-idf and then encrypt it and write it to the flash, it works perfectly and I get this message:

I (0) cpu_start: App cpu up.
I (145) cpu_start: Pro cpu start user code
I (145) cpu_start: cpu freq: 160000000
I (145) cpu_start: Application information:
I (150) cpu_start: Project name:     main
I (154) cpu_start: App version:      1
I (159) cpu_start: Compile time:     Aug 27 2021 14:32:23
I (165) cpu_start: ELF file SHA256:  c81c9b73a2a94ecd...
I (171) cpu_start: ESP-IDF:          v4.3-dirty
I (176) heap_init: Initializing. RAM available for dynamic allocation:
I (183) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (189) heap_init: At 3FFB3840 len 0002C7C0 (177 KiB): DRAM
I (196) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (202) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (208) heap_init: At 4008BB88 len 00014478 (81 KiB): IRAM
I (216) spi_flash: detected chip: generic
I (219) spi_flash: flash io: dio
W (224) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (233) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (261) nvs: NVS partition "nvs" is encrypted.
�EEPROM is Empty
EEPROM was initialised
SPIFFS mounted
Flash encryption enabled

but when I compile it with Arduino-IDE and then encrypt it and write it to the app partition, I got this error.

entry 0x400805e8
�failed to initialise EEPROM
EEPROM is Empty
failed to initialise EEPROM
EEPROM is Empty
E (4004) SPIFFS: spiffs partition could not be found
An Error has occurred while mounting SPIFFS
Flash encryption enabled

can anyone help about this issue?

### Sketch:  
#include <Arduino.h>
#include <base64.h>    // for creating the "_auth"-String (64bit) 
#include "Wire.h"       // for I2C-connection 
#include "EEPROM.h"
#include "ETH.h"
#include "SPIFFS.h"
#include "esp_flash_encrypt.h"

// ########################################################################################## 

// static IP 
IPAddress myip(192, 168, 1, 10);    // Static IP for ESP32
IPAddress gwip(192, 168, 1, 1);     // Gateway & Accesspoint IP 
IPAddress mask(255, 255, 255, 0);   // Subnetmask 
IPAddress myDNS(192, 168, 1, 1);      // optional 

// ########################################################################################## 
// Setup default values for eeprom 

struct Storage{
  // WiFi Data (Client) currently not used 
  char ssid[24]        = "SSID";                  // WiFi network SSID 
  char password[24]    = "PASSWORD";            // WiFi network PW 
  unsigned long timeoutRouter   = 65;               // Time (seconds) to wait for WiFi access, after that own WiFi Access Point starts  
  
  // Eth Data (Client) 
  unsigned long timeoutEthernet = 65;               // Time (seconds) to wait for ETH access OPTION: after that own WiFi Access Point could be started  
  char esp32HostnameETH[20]   = "ESP32";  // Hostname of the ESP32 in the Network 
  
  // Ntrip Caster Data 
  char host[40]           = "192.168.1.1";      // Main
  char host_back[40]      = "192.168.1.2";      // Backup  

  // Network-Mode  
  byte enableNetwork = 0;                           // 0 = ETH only. No WiFi activated. 
                                         
  long baudOut = 115200;                  // Baudrate of RTCM Port 

}; Storage Settings;

// ------------------------------------------------------------------------------------------ 
// EEPROM Data Handling 
// ------------------------------------------------------------------------------------------ 
#define EEPROM_SIZE 512
#define EE_ident1 0xED  // Marker Byte 0 + 1
#define EE_ident2 0xEF

// ------------------------------------------------------------------------------------------ 

byte EEprom_empty_check() {
  byte result = 0;
  if (!EEPROM.begin(EEPROM_SIZE)) {
    Serial.println("failed to initialise EEPROM"); delay(1000);
    result = 0;
    }
  
  if (EEPROM.read(0)!= EE_ident1 || EEPROM.read(1)!= EE_ident2) {
    result = 1;  // is empty 
    Serial.println("EEPROM is Empty"); delay(1000);
  }
  
  if (EEPROM.read(0) == EE_ident1 && EEPROM.read(1) == EE_ident2) {
    result = 2;     // data available 
    Serial.println("EEPROM was initialised"); delay(1000);
  }
  return result; 
} // byte EEprom_empty_check() 

// ------------------------------------------------------------------------------------------ 
void EEprom_write_all() {  // called if EEPROM empty 
  EEPROM.write(0, EE_ident1);
  EEPROM.write(1, EE_ident2);
  EEPROM.write(2, 0); // reset Restart blocker 
  EEPROM.put(3, Settings);
  EEPROM.commit();
} // void EEprom_write_all() 

// ------------------------------------------------------------------------------------------ 
void EEprom_read_all() {
  EEPROM.get(3, Settings);
} // void EEprom_read_all() 

// ------------------------------------------------------------------------------------------ 
void EEprom_show_memory() {
  byte c2 = 0, data_;
  Serial.println(EEPROM_SIZE);
  Serial.println(" bytes read from Flash . Values are:\n");
  
  for (int i = 0; i < EEPROM_SIZE; i++) { 
    data_=byte(EEPROM.read(i));
    if (data_ < 0x10) Serial.print("0");
    Serial.println(data_,HEX); 
    if (c2 == 15) {
      Serial.println(" ");
    }
    else if (c2>=31) {
      Serial.println(""); // NL 
      c2 =- 1;
    }
    else Serial.println(" ");
    c2++;
  }
} // void EEprom_show_memory() 

// ------------------------------------------------------------------------------------------ 
// Restore EEprom Data 
// ------------------------------------------------------------------------------------------ 
void restoreEEprom() {

  if (EEprom_empty_check() == 1) { // first start? 
    EEprom_write_all();     // write default data 
  }
  if (EEprom_empty_check() == 2) { // data available 
    EEprom_read_all();
  }
  //EEprom_show_memory();
} // void restoreEEprom() 

// ########################################################################################## 
// Main-setup procedure 
void setup() {
  // First port to be opened: Serial0 for debugging (USB) 
  Serial.begin(115200, SERIAL_8N1, RX, TX); // Serial0 (USB) for debug-messaging 

    restoreEEprom(); // Initializing eeprom 

  // Initialize SPIFFS to loading files from flash
  if(!SPIFFS.begin(true)){
    Serial.println("An Error has occurred while mounting SPIFFS");
  }
  else
    Serial.println("SPIFFS mounted");

} // void setup() 

// ########################################################################################## 
void loop() {

  if (esp_flash_encryption_enabled() == true) {
    Serial.println("Flash encryption enabled\n");
  } else { Serial.println("Flash encryption disabled\n"); }
  delay(10000);
} // void loop()

chegewara, 2021-08-30:
https://github.com/espressif/arduino-esp32/blob/master/LICENSE.md

> Primarily used for software libraries, the GNU LGPL requires that derived works be licensed under the same license, but works that only link to it do not fall under this restriction.

Mos135, 2021-09-01:
> https://github.com/espressif/arduino-esp32/blob/master/LICENSE.md
> 
> > Primarily used for software libraries, the GNU LGPL requires that derived works be licensed under the same license, but works that only link to it do not fall under this restriction.

I did not understand your meaning!

VojtechBartoska, 2022-03-30:
Hello @Mos135, I'm not sure what exactly is this request about. We will take a look on it with the Team and let you know later.

VojtechBartoska, 2022-04-05:
@Mos135 After discussion of this feature request we decide that it's not doable. I'm closing it, thanks for your contribution!

closed 2022-04-05 by VojtechBartoska